adafruit/circuitpython #5: Does not compile for BOARD=feather_m0_bluefruit_le

turbinenreiter, 2016-09-28:
```
$ make CROSS=1 BOARD=feather_m0_bluefruit_le
CC ../py/../extmod/vfs_fat_file.c
../py/../extmod/vfs_fat_file.c:257:13: error: 'MP_QSTR_FileIO' undeclared here (not in a function)
     .name = MP_QSTR_FileIO,
             ^
../py/mkrules.mk:47: recipe for target 'build-feather_m0_bluefruit_le/py/../extmod/vfs_fat_file.o' failed
make: *** [build-feather_m0_bluefruit_le/py/../extmod/vfs_fat_file.o] Error 1
```

tannewt, 2016-09-28:
What gcc are you using?

Please try cleaning and rebuilding. Sometimes the qstr generation isn't
retriggered.

turbinenreiter, 2016-09-28:
I swear I cleaned and rebuilt before and it didn't work, but now it did. Thanks.

closed 2016-09-28 by turbinenreiter

tannewt, 2016-09-28:
No problem! Let me know how it goes.