adafruit/circuitpython #19: I2C (seemingly) doesn't work

turbinenreiter, 2016-10-17:
`from machine import I2C, Pin

i2c_bus = I2C(scl=Pin.board.SCL, sda=Pin.board.SDA, freq=400000)
print(i2c_bus.scan())`

A scan of the I2C bus always turns up empty, although there are I2C devices connected, as confirmed with other hardware that finds the devices.

tannewt, 2016-10-17:
Hey, thanks for filing an issue. I changed the I2C and SPI classes a little bit to make it more explicit when the actual hardware is dedicated to the function.

Try:
```
from machine import I2C, Pin

with I2C(scl=Pin.board.SCL, sda=Pin.board.SDA, freq=400000) as i2c_bus:
  print(i2c_bus.scan())
```

The `with` statement implicitly calls `init` and `deinit` to turn on and off the underlying hardware. This way if you restart or want to use those pins for something else later they are properly deinitialized. I suggest using the `with` format above because it'll still call `deinit` after a keyboard interruption. If you want to manage it yourself you can also do:

```
from machine import I2C, Pin

i2c_bus = I2C(scl=Pin.board.SCL, sda=Pin.board.SDA, freq=400000)
i2c_bus.init()
print(i2c_bus.scan())
i2c_bus.deinit()
```

Beware that this will leave the i2c hardware in a bad state if you ctrl-c during the scan. I thought about making the constructor always reset the hardware so it can claim it but didn't want to risk silently changing hardware functionality if two uses overlap.

Please try the examples and close the issue if it does work. Thanks!

turbinenreiter, 2016-10-18:
Thank's, now it works.

closed 2016-10-18 by turbinenreiter

tannewt, 2016-10-18:
Perfect! Sorry I didn't make that change clearer. I'm cleaning up some docs today and that should provide a better way to understand the API changes I'm making.