adafruit/circuitpython #22: Stop using machine in favor of microcontroller.

tannewt, 2016-10-17:
MicroPython's `machine` module varies a lot between different ports. Instead, we should define a separate API `microcontroller` in `shared-bindings` that ensures identical APIs by sharing code. It should be divided into three submodules:

* `microcontroller.core` Processor related functionality such as sleep states.
* `microcontroller.pin` Microcontroller level pin definitions such as PA13. Board names should live elsewhere. (In a separate `board` for example.)
* `microcontroller.peripheral` Provides classes for common hardware peripherals such as GPIO, I2C, SPI, Timers, UART, DAC and ADC.

deshipu, 2016-10-20:
https://imgs.xkcd.com/comics/standards.png

tannewt, 2016-11-29:
`microcontroller`, `nativeio` and `board` have been added in place of `machine`.

closed 2016-11-29 by tannewt