adafruit/circuitpython #28: Explain when main is running on connect

tannewt, 2016-10-21:
When you plug in the board and main is running make sure that it explains whats going on.

tdicola, 2016-10-21:
Let's be careful to also support scenarios where it's a script and not a user connecting to the serial output.  Like a script to read sensor readings that are periodically taken by the micropython board.  It would cause problems to suddenly start getting output like 'main.py is running'.  Let's put some kind of global config or function to call to disable the output in main.py or boot.py (see the esp8266 os.debug function for an example of enable/disable debug output--this is kinda similar).

tannewt, 2017-06-18:
I'm going to close this. Its tricky to detect when USB is actually plugged in versus when the first character is received. That makes it hard to do right.

closed 2017-06-18 by tannewt