adafruit/circuitpython #33: build broken for feather_m0_basic

g-ollivier, 2016-10-22:
Hello,
Build works for default arduino_zero board, but it doest not work anymore for feather_m0_basic

In file included from access_vfs.h:33:0,
                 from access_vfs.c:29:
asf/common/services/storage/ctrl_access/ctrl_access.h:142:25: fatal error: rom_fs.h: No such file or directory
   #include LUN_0_INCLUDE
                         ^
compilation terminated.
../py/mkrules.mk:47: recipe for target 'build-feather_m0_basic/access_vfs.o' failed

I tried to get arround with a symbolic link between access_vfs.h and rom_fs.h but I get more error messages.

(This seems linked to PR 31 https://github.com/adafruit/micropython/pull/31)

g-ollivier, 2016-10-22:
After update of boards/feather_m0_basic/conf_access.h with similar parameters as for arduino_zero, builds is ok

#define LUN_0_INCLUDE                           "access_vfs.h"
#define Lun_0_test_unit_ready                   vfs_fs_test_unit_ready
#define Lun_0_read_capacit y                     vfs_fs_read_capacity
#define Lun_0_wr_protect                          vfs_wr_protect
#define Lun_0_removal                              vfs_removal
#define Lun_0_usb_read_10                       vfs_usb_read_10
#define Lun_0_usb_write_10                      vfs_usb_write_10
#define LUN_0_NAME                              "\"MicroPython VFS[0]\""

tannewt, 2016-10-24:
Thanks for the heads up. Sorry I broke the build! I've setup Travis to check the build against pushes and pull requests so we can catch it earlier.

I've fixed the build in 16584dfee6ec09a.

closed 2016-10-24 by tannewt

tannewt, 2016-10-25:
Thanks for the report! I'll fix it on Monday.