adafruit/circuitpython #35: Add descriptive help() function

tdicola, 2016-10-24:
Currently help() is unimplemented, but we should add a descriptive block of text that explains at least the micropython firmware version.  Ideally something with instructions on getting started, like with the esp8266 and microbit ports, would be nice to add too.  See esp8266 help implementation: https://github.com/adafruit/micropython/blob/master/esp8266/help.c

tannewt, 2016-12-08:
What should we have the `help()` provide? Ideally it'd be short because it will be stored in flash memory.

tannewt, 2016-12-09:
Add a link to adafruit.com/micropython and autogenerate lists of available modules and their members.

tannewt, 2016-12-20:
Fixed in 427d33c706236985dff950382e1c6b896b36b1bd. Example:

```
>>> help()
Welcome to Adafruit MicroPython v1.8.6-183-ge8db23d-dirty!

Please visit learn.adafruit.com/category/micropython for project guides.

Built in modules:
        __main__
        builtins
        micropython
        array
        ucollections
        ustruct
        math
        sys
        gc
        urandom
        framebuf
        microcontroller
        nativeio
        board
        uos
        time
        neopixel_write
        samd
        bitbangio
>>>
```

closed 2016-12-20 by tannewt