adafruit/circuitpython #37: v1.8.5-20161020-13-gf5ca668 experimental build boot.py and file writing

ptorrone, 2016-10-25:
this is for the experimental build (v1.8.5-20161020-13-gf5ca668 on 2016-10-24', machine='Adafruit Metro M0 with Flash (Experimental) with samd21g18)

boot.py does not load on start and writing to the file system on mac 10.11.6 (usb drive MICROPYTHON) gives an error, but does save (see screenshot).
.................
Traceback (most recent call last):
 File "<stdin>", line 1, in <module>
NameError: name 'help' is not defined
import os
os.uname()
(sysname='samd21', nodename='samd21', release='1.8.5', version='v1.8.5-20161020-13-gf5ca668 on 2016-10-24', machine='Adafruit Metro M0 with Flash (Experimental) with samd21g18')
...............

[image]

tannewt, 2016-10-25:
What program were you using to edit the file? It looks like its configured to save multiple versions of the same file.

Could you please do a soft reset (CTRL-D) and paste the output here? Output of "os.listdir()" would be welcome too. Thanks!

ptorrone, 2016-10-26:
i was using a text edit.

interesting, after a soft reboot it runs boot.py and main.py so that "fixes" it.

>>> 
soft reboot
Traceback (most recent call last):
  File "main.py", line 7, in <module>
KeyboardInterrupt: 
Adafruit MicroPython v1.8.5-20161020-13-gf5ca668 on 2016-10-24; Adafruit Metro M0 with Flash (Experimental) with samd21g18
>>> help
'Welcome to Adafruit Micropython!!!!'
>>> os.listdir()
['._.Trashes', '.TemporaryItems', 'boot.py', '._boot.py', '.fseventsd', '.Trashes', 'main.py', '._.TemporaryItems']
>>>

tannewt, 2016-11-29:
I think this is fixed now.

closed 2016-11-29 by tannewt