adafruit/circuitpython #39: import errors on feather_m0_basic

g-ollivier, 2016-10-27:
Hello,
With release v1.8.5-20161020, on feather_m0_basic, commands like import os or import time are running.
But with latest build, it does not work.

tannewt, 2016-10-27:
Use uos and utime. They feature extra methods that the CPython variants
don't have. I want that to be clear with the import.

g-ollivier, 2016-10-27:
Ok - I modified ssd1306 module so that it uses utime instead of time and it works.
So closing this issue.
Thank you

closed 2016-10-27 by g-ollivier

tannewt, 2016-10-27:
Sorry I was on my phone and I want to give a little more reasoning behind it.

First, I want to make modules have consistent methods across ports and between micropython and cpython. That way people's knowledge can be applied in all of those places. Adding extra methods like utime and uos are not compatible with cpython time and os so I want to make it clear in imports what functionality is expected. That way, when moving code or knowledge from one to the other its clear where it applies.

Second, I do intend on adding time and os back without the micropython specific methods. Next time I'll wait to remove the aliasing until I have those too although they may be shipped as separate mpy files.

Thanks again for all of your reports. I'm really trying to push hard in evolving this quickly so we can stabilize APIs sooner rather than later.