adafruit/circuitpython #40: need a spi flash formatter

ptorrone, 2016-10-27:
hi hi! need a spi flash formatter to start over with a clean file system in case something goes really wrong (testing on macs with autosaving on for text edit for example!)

tannewt, 2016-12-01:
Plan is that for internal flash, provide two binaries, one without the flash system and one with it.

For external flash, the bootloader is responsible for keeping the flash FS valid.

tannewt, 2016-12-01:
Look into Etcher for reflashing the spi flash.

tannewt, 2016-12-07:
I'm gonna take this off the milestones because some firmware flashing stuff may make this unnecessary.

tannewt, 2016-12-20:
We can provide a separate binary people can flash to format it if they can't do it directly over mass storage.

closed 2016-12-20 by tannewt