adafruit/circuitpython #44: i2c but init on soft reboot

ladyada, 2016-10-30:
a collision between https://github.com/adafruit/micropython/issues/29 & auto-reset
if your main.py uses i2c it can't be re-run/saved because i2c will fail. 

(i ~think~ this is also causing main.py with i2c in it to fail on first run but its impossible to tell cuz i can't see main.py error output)
  
```
>>> i2c = machine.I2C(machine.Pin('SCL'), machine.Pin('SDA'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: I2C bus init error
```

tdicola, 2016-10-30:
Yeah this is currently by design, but #29 will fix it.  For now use a context manager syntax in main.py like I show in this video and example: https://youtu.be/lbd9M7fY4mw?list=PLjF7R1fz_OOXrI15wuXeESA0aA4VzcWSi&t=1751 and https://gist.github.com/tdicola/2c42bfe534838b3c159c1b3cc3e58ac8  Also an example in the SAMD21 guide: https://learn.adafruit.com/micropython-for-samd21/quick-start-guide#i2c-devices

closed 2016-10-30 by tdicola

ladyada, 2016-10-30:
aww yah that helps a ton!