adafruit/circuitpython #45: utime.sleep() in main.py can "brick"

ladyada, 2016-10-30:
i momentarily forgot that utime.sleep() takes seconds instead of milliseconds and called utime.sleep(5000) in main.py and now i can't ^C to the REPL (COM port does appear tho), ^D, or get the mass storage to appear. yay!

probably some bug? who knows :) imma reflash this feather m0 to clear out the storage - but if this were a metro m0 you wouldn't be able to, so having some sort of external tool to clear out the SPI flash could be handy (e.g. https://github.com/adafruit/micropython/issues/40)

tannewt, 2016-10-30:
Yup I have a todo to check that.

microbuilder, 2016-10-30:
We have a factory reset pin on the Bluefruit modules and we check the pin state in the bootloader to see if we should clear all the config settings, etc., to factory defaults.  That's saved our butts more than a few times and might be worth thinking about here? 

We also implemented a feature where holding the DFU button down for 5s causes a factory reset on release (with a distinct blinky pattern to let you know when you can release).

Maybe neither of those are relevant here, but figured I'd toss them out. A factory reset solder jumper/pad on the bottom might be a good idea just as a backup plan though, so you can still just connect a wire by hand to that and GND when you start the module up with no soldering.

ladyada, 2016-10-30:
hmm, on feather m0 we don't have a userland button, and not sure the metro m0 will have one either (but it could be added!)

one option, maybe shorting SWDIO to ground while powering up does a factory reset? as long as the firmware set a pullup on, waited a few ms for the voltage to stabilize, before checking? SWDIO is available as a pad but not a true 'pin' so its unlikely a user will connect to it

tannewt, 2016-10-31:
I don't think we actually need a pin. I can have the delay loops check for ctrl-c when the USB is plugged in. Right now its just the python interpreter doing it so if you get stuck in C code you can't ctrl-c until it finishes.

ladyada, 2016-10-31:
if you utime.delay() it also seems to kill mass storage, does being stuck in C code also halt mass storage handling? (cuz oetherwise i could have just deleted main.py)

tannewt, 2016-10-31:
What version are you running? It should have been fixed on Friday or so.
https://github.com/adafruit/micropython/commit/86a1d1db3004ce3c3d052e1e54941dfa4c369de7

closed 2016-10-31 by tannewt

ladyada, 2016-10-31:
cant recall (its at home) but will try again with fresh version!

tannewt, 2016-10-31:
cool cool!

commit 19e5f1fe (2016-10-31): atmel-samd: Break out of delays with CTRL-C.

This means you can CTRL from within time.sleep() and that autoreset
will work from within a long sleep too. Fixes #45.