adafruit/circuitpython #47: Investigate cookiecutter templates for easy library generation

tdicola, 2016-10-31:
Putting this on the main micropython lib but this really applies to the libraries (micropython-adafruit-* repos).  We're starting to get a lot of conventions for libraries, like having a .travis.yml to support auto MPY generation or adding libraries as submodules to the micropython bundler repo.  Let's see if we can capture all these things in a adafruit micropython library template so that a command can be run to give the scaffolding for a new library.  This could include stuff like .travis.yml, a readme with instructions on exactly what to do to get bundler support, an examples folder, etc.

willingc, 2016-11-03:
@tdicola Cookiecutter is an awesome project. We are using it for Project Jupyter for widget constructions. One particularly nice cookiecutter template is https://github.com/audreyr/cookiecutter-pypackage cc/ @audreyr @hackebrot

hackebrot, 2016-11-06:
@willingc I do think [cookiecutter](https://github.com/audreyr/cookiecutter) is a great project (but I'm a little biased 😁)

tannewt, 2017-01-19:
Here is an initial cookiecutter for CircuitPython libraries: https://github.com/adafruit/cookiecutter-adafruit-circuitpython

closed 2017-01-19 by tannewt

willingc, 2017-01-19:
Nice! Thanks @tannewt.