adafruit/circuitpython #52: atmel-samd: Can't save to drive from Mac's built in text editor

tdicola, 2016-11-11:
Trying to save a main.py I get an error from the built in text editor that the drive can't be found or the file can't be saved:
[image]

PT & Pedro have noticed this too.  Perhaps something is up with the drive and how text edit expects the filesystem, etc. to work.  Needs a bit more investigation.

willingc, 2016-11-11:
@tdicola Is this possibly a SD formatting issue?

tannewt, 2016-11-30:
I think this was fixed as part of ccbb5e84f980e252f7380d96fcc5c417a8a64523. There was a bug where the ram cache got deleted by the gc and then a write to flash was attempted from it. Another bug squashed was an infinite loop when failing to allocate the ram cache.

closed 2016-11-30 by tannewt