adafruit/circuitpython #53: atmel-samd: ILI9341 TFT driver locks up device when attempting hardware SPI calls

tdicola, 2016-11-16:
Trying to use the TFT FeatherWing with atmel-samd from the current master but am noticing during its initialization it appears get stuck and lock up (not even ctrl-c saves it).  Connect the TFT FeatherWing to the board and make sure the FeatherWing on switch is flipped on.  Then load the latest ili9341.mpy and rgb.mpy from the releases here: https://github.com/adafruit/micropython-adafruit-rgb-display/releases  Try the following code:
```
import machine
import ili9341
spi = machine.SPI(baudrate=32000000, clock=machine.Pin('SCK'), MOSI=machine.Pin('MOSI'), MISO=machine.Pin('MISO'))
display = ili9341.ILI9341(spi, cs=machine.Pin('D9'), dc=machine.Pin('D10'))
```
You'll see the display creation never finishes and locks up the board.  I have a feeling it's getting stuck during its SPI calls to initialize the display (the display MISO is never actually used, it's only MOSI with the board sending the display data).

tannewt, 2016-11-30:
What feather was this with? Did it have SPI flash?

tdicola, 2016-12-02:
No external flash, this was with the Feather M0 proto board.

tannewt, 2016-12-05:
Fixed by 26229efe78f6c5f8598d6f9f1b2261c5a84d7a7b. I tested the ili9341 on both a Feather M0 Bluefruit LE and Feather HUZZAH esp8266. Note, it requires the latest drivers including the bus device stuff.

closed 2016-12-05 by tannewt