adafruit/circuitpython #54: Some GPIO not accessible

bobricius, 2016-11-22:
Hi, some special ZERO GPIO are not accessible.

etc
 * | 25         |                  |  PB03  | RX
 * | 26         |                  |  PA27  | TX 
 * | 27         |                  |  PA28  | USB_HOST_ENABLE
 * | 42         | AREF         |  PA03  | 
 * | 30         |                  |  PB22  | EDBG_UART TX 
 * | 31         |                  |  PB23  | EDBG_UART RX
 * | 32         |                  |  PA22  | EDBG_SDA
 * | 33         |                  |  PA23  | EDBG_SCL

tannewt, 2016-11-22:
What port and what version (commit)? I can look tomorrow.

bobricius, 2016-11-22:
etc now ... micropython/atmel-samd/boards/arduino_zero/pins.c

This pins are not defined and maybe more, compared to arduino samd core variant
I understand that this pins are usually not used.

tannewt, 2016-11-22:
The pins should be available through their microcontroller names such as PA22 from the `microcontroller.pin` module. That module's pins are done automatically based on defines. https://github.com/adafruit/micropython/blob/master/atmel-samd/common-hal/microcontroller/__init__.c#L148

tannewt, 2016-11-30:
Having these accessible through `microntroller.pin` is enough to me.

closed 2016-11-30 by tannewt