adafruit/circuitpython #56: Add built-in neopixel library

tannewt, 2016-12-01:
So that they can't delete it off the file system.

tannewt, 2017-08-08:
We should do this as a submodule with a script to autoupdate similar to the bundle. We should also prefer filesystem copies to built in versions in case there are bug fixes.

ladyada, 2017-08-08:
maybe only built in on cplay express? im ok with that!

tannewt, 2017-08-08:
I was thinking all of the Express boards since they have the room.

dhalbert, 2017-08-08:
Add as a git submodule and build in as an internal frozen module? Disadvantage is you need a new CPy release to update. Advantage is built-in, can't be deleted, and no RAM used for the code when you import. You could add more convenience functions without worrying about RAM usage.

tannewt, 2017-08-08:
Yeah, that was my plan. We can get around the upgrade disadvantage by preferring modules from the filesystem first.

tannewt, 2017-08-10:
Note that frozen modules are currently loaded when the path is "". I think we should change it to something like ".frozen" to be explicit. We'll need to change the make-frozen.py script to change the filenames.

dhalbert, 2017-08-18:
Working on branch https://github.com/adafruit/circuitpython/tree/issue_56_frozen_neopix_lib; checked in snapshot.

Modules to freeze will be added as git submodules to the `frozen/` directory. I modified the makefiles to allow multiple directories to freeze: `FROZEN_MPY_DIRS` instead of singular `FROZEN_MPY_DIR`. Frozen modules to include for a specified board are specified in `mpconfigboard.mk` files. Rewrote frozen code generation rules in `py/mkrules.mk`; I think it's more straightforward now. There's still the peculiar `$(BUILD)/$(BUILD)/frozen_mpy.o` path in `py/py.mk`. I haven't gotten rid of this successfully.

Tested on CPX: only one neopixel is working. Not sure if that's a library issue or a frozen module issue.

commit 74cfdeb3 (2017-08-23): Added ability to freeze multiple directories; freeze neopixel library in cpx build (#199)

Reworked frozen module support: clean up makefiles and handle multiple directories.
Modules to freeze are included as git submodules.

Add neopixel to circuitplayground express build.

Fixes #56
closed 2017-08-23 by tannewt