adafruit/circuitpython #57: esp8266: Ctrl-c during time.sleep resets the board

tdicola, 2016-12-02:
Odd thing I noticed, perhaps not related to our firmware but worth looking at further, if you press Ctrl-C during a time.sleep operation the board does a soft reset.

To repro, in the REPL run:
```
import time
time.sleep(5)
```
Then press Ctrl-C and you'll see the board resets instead of breaking out of the sleep:
```
>>> import time
>>> time.sleep(5)
PYB: soft reboot
#13 ets_task(40100164, 3, 3fff8408, 4)
Adafruit MicroPython v1.8.6-147-gd2aa05a on 2016-11-28; ESP module with ESP8266
>>> 
```

tdicola, 2016-12-02:
Hrm it looks like this works fine with upstream esp8266, here's the same behavior on micropython.org 1.8.6:
```
>>> import time
>>> time.sleep(5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt: 
>>> 
```

Perhaps when the exception handler runs some state to remember to kick the watchdog isn't firing and the ESP8266 resets itself?

dhalbert, 2017-09-06:
This bug came back as of 2.0.0-rc1 or earlier. Also see the same thing with
```
while True: pass
```
then type ctrl-C. Board will soft-reset.

dhalbert, 2017-09-06:
Fixed by #241

commit f2c787d0 (2017-02-15): Fix the keyboard exception in REPL by ignoring forced exit results when in REPL. Also, print the exception correctly. Fixes #57
closed 2017-09-06 by dhalbert