adafruit/circuitpython #65: Support SPI bus with only MOSI or MISO line

tdicola, 2016-12-12:
Right now the nativeio.SPI class seems to require both MOSI and MISO to be specified, for example on esp8266:
```
>>> import nativeio
>>> import board
>>> spi = nativeio.SPI(clock=board.SCK, MOSI=board.MOSI)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Pins not valid for SPI
>>> 
```
Compared to specifying both MOSI and MISO which does work fine:
```
>>> spi = nativeio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
>>>
```
Talking earlier we do think supporting just a MOSI or MISO makes sense and this might be a small bug.

tannewt, 2016-12-12:
OK, my guess is that it only works on atmel-samd. I'll have to look at it
for esp. Thanks!

deshipu, 2017-02-11:
You can disable the miso/mosi pins in the ESP8266's SPI peripheral, but currently the flags that do that are hardcoded in the HSPI functions -- you would need to change that.

See for example https://github.com/adafruit/circuitpython/blob/master/esp8266/hspi.c#L195

commit b7ded4c9 (2017-02-19): esp8266: Allow for MOSI or MISO only SPI. Fixes #65
closed 2017-02-19 by tannewt