adafruit/circuitpython #67: atmel-samd: Filesystem out of free space when copying over empty files.

tdicola, 2017-01-03:
I noticed something odd using the samd21 port on Feather M0 basic, it seems like empty files (and perhaps even files below 4k in size) are counting as larger size files.  I can only copy over 10 0 byte files before I receive an out of free space error.

Version:
```
Adafruit CircuitPython v1.8.6-187-g3986870-dirty on 2016-12-21; Adafruit Feather M0 Basic with samd21g18
```

Repro steps:
- Flash firmware above (latest build) to ensure filesystem is empty.
- Create a bunch of zero byte files, like:
  ```
  touch empty_file_1.txt
  touch empty_file_2.txt
  etc...
  ```
- Attempt to drag a large number of empty files (20) to the board in OSX finder.  Notice the copy fails after 10 files:
[image]
- As another check, try to create a file using the serial REPL and python code.  Notice trying to open a file for writing fails with an OSError 30 (not enough free space?):
  ```
  >>> foo = open('foo.txt', 'w')
  Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
  OSError: 30
  >>>
  ```

tdicola, 2017-01-03:
Interestingly the uos.stat command sees the files as 0 bytes (the 4th to last tuple entry, a 0 below):
```
>>> uos.stat('empty_file_8.txt')
(32768, 0, 0, 0, 0, 0, 0, 536706658, 536706658, 536706658)
```
But trying to copy more files over still results in an out of space error in OSX finder.

tannewt, 2017-01-03:
Can you copy more files over if some of them are in a directory? You can
fill up a file system with empty files because of the overhead. I'm
surprised its happening at ten files though.

tannewt, 2017-01-04:
Copying via command line works fine. Mac OS X is notorious for lots of spare files so it looks unique to Finder. Nothing on the MicroPython side we can do about it.

closed 2017-01-04 by tannewt