adafruit/circuitpython #71: esp8266: cannot set RTC hour field; API is confusing

cversek, 2017-01-10:
```python
Adafruit CircuitPython v1.8.5-20161020-234-g5b7fcf6-dirty on 2017-01-09; ESP module with ESP8266
>>> 
>>> import utime, machine
>>> rtc = machine.RTC()
>>> rtc.
datetime        memory          alarm           alarm_left
irq             ALARM0
>>> rtc.datetime((2000,1,1,0,0,0,0,0))
>>> rtc.datetime()
(2000, 1, 1, 5, 0, 0, 15, 838)
>>> rtc.datetime((2000,1,1,6,0,0,0,0))
>>> rtc.datetime()
(2000, 1, 1, 5, 0, 0, 4, 205)
>>> rtc.datetime((2000,1,1,4,0,0,0,0))
>>> rtc.datetime()
(2000, 1, 1, 5, 0, 0, 2, 779)
>>> 
```
The API for the RTC class doesn't match [latest micropython API](http://docs.micropython.org/en/v1.8.6/esp8266/library/machine.RTC.html) - which is OK if you want to be different, but the [documentation](http://circuitpython.readthedocs.io/en/latest/docs/library/machine.RTC.html) doesn't match the methods any more. I suppose that the `RTC.datetime` method called with a datetime tuple is like the old `RTC.init` method; however, I can't make sense of the behavior (see code above) - what I suppose to be the `hour` field is stuck at a weird number.  What do these `datetime` fields even mean...especially the dt[4] entry? Please tell me I'm not going crazy.  I will probably have to revert back to when circuitpython was micropython and before all the time APIs started changing - if I can't get time keeping working without nasty surprises.  Sorry to be that way.  I love micropython, the Feather HUZZAH, and what Adafruit is trying to do here in general.

tannewt, 2017-01-10:
Is it the onboard rtc? I'll take a look tomorrow. I'm headed to bed now.
I've only done external rtc stuff thus far. I don't expect machine to be
any different from upstream. Only native io and the drivers based on it are
different.

Thanks for trying the early beta and filing an issue. We'll get the kinks
worked out before 1.0. :-)

cversek, 2017-01-10:
Yes it's the internal RTC.  No problem, I am having fun despite the bugs.  I will check the behavior tomorrow night on the micropython master build and the one from the vagrant machine from Tony Ds guide if you haven't cracked it by then.

tannewt, 2017-01-10:
It looks like the docs are wrong in both places (because we haven't changed this at all.)

The code that implements it is here: https://github.com/adafruit/circuitpython/blob/master/esp8266/machine_rtc.c#L137

The expected field order there is: year, month, day, weekday (0-6), hour, min, second, millisecond. Setting the RTC ignores the weekday which explains why you changing index 3 doesn't change anything.

I also have no idea why this is the case because its not the order of upstream's struct_time. https://docs.python.org/3.4/library/time.html#time.struct_time

I hope that makes sense.

cversek, 2017-01-11:
Thanks, that actually makes a lot of sense and explains these curious [lines of code](https://github.com/micropython/micropython/blob/master/esp8266/scripts/ntptime.py#L33) in the ntptime.py script that was the source of my confusion:
```python
tm = utime.localtime(t)
tm = tm[0:3] + (0,) + tm[3:6] + (0,)
machine.RTC().datetime(tm)
```
I just started digging into the C-extension module code and I should have probably looked there before complaining.  Ahh.... what confusion a couple simple in-line comment could have saved.  May I suggest
```python
tm = utime.localtime(t)             #(year, month, day, weekday (0-6), hour, min, second, millisecond)
tm = tm[0:3] + (0,) + tm[3:6] + (0,)
machine.RTC().datetime(tm)          #(year, month, day, None, hour, min, second, None)
``` 
?
Especially, since there are no doc-strings that I know of in micropython and I heard that comments are stripped out when the scripts directory is compiled to byte code (right?).  API consistency is an issue that I am probably not qualified to raise ;)

closed 2017-01-11 by cversek