adafruit/circuitpython #72: Freeze after dir.

tannewt, 2017-01-10:
From the [forum](https://forums.adafruit.com/viewtopic.php?f=60&t=109553#p547448):

Install circuitPython on Zero Pro.
It worked.
And I try the code.

```python
import nativeio as io
import board as pin
led = io.DigitalInOut (pin.D13)

dir (led.direction)
```

Freeze....

tannewt, 2017-01-10:
This works just fine for me.

```
Adafruit CircuitPython 0.8.0 on 2017-01-09; Arduino Zero with samd21g18
>>> import nativeio as io
>>> import board as pin
>>> led = io.DigitalInOut(pin.D13)
>>> dir(led.direction)
['in', 'out']
```

Is there anything else on the flash file system or code that was run before?

willingc, 2017-01-11:
Wondering if the whitespace in `io.DigitalInOut(pin.D13)` is causing `led` to not set correctly. Though you would think it would throw an error if so.

sugarflower, 2017-01-11:
dir(pin.D13)
[]

I see.

tannewt, 2017-01-11:
Thanks for the idea @willingc. I think it was a flash error. The user reported back that it works now: https://forums.adafruit.com/viewtopic.php?f=60&t=109553&start=15#p547626

closed 2017-01-11 by tannewt