adafruit/circuitpython #77: esp8266: machine.I2C broken

cversek, 2017-01-17:
```python
Adafruit CircuitPython v1.8.5-20161020-560-gbc92545-dirty on 2017-01-17; ESP module with ESP8266
>>> import machine
>>> machine.I2C(machine.Pin(5),machine.Pin(4))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid I2C peripheral
```
I believe the exception is generated in `extmod/machine_i2c.c` line [297](https://github.com/adafruit/circuitpython/blob/master/extmod/machine_i2c.c#L297).

tannewt, 2017-01-17:
That isn't `machine`'s API. Its `machine.I2C(-1, scl=machine.Pin(5), sda=machine.Pin(4))`. The `-1` tells it to use bitbanging.

I'd recommend using `bitbangio` instead because its [well documented](https://circuitpython.readthedocs.io/en/latest/shared-bindings/bitbangio/__init__.html) and shares the same API as `nativeio.I2C` which is used on the SAMD. It requires locking though in case other code is talking on the same device. The [adafruit_bus_device](https://github.com/adafruit/Adafruit_CircuitPython_BusDevice) ([docs](https://circuitpython.readthedocs.io/projects/bus_device/en/latest/adafruit_bus_device/index.html#module-adafruit_bus_device.i2c_device)) library was created to make it easy to get locking and settings right.

closed 2017-01-17 by tannewt

cversek, 2017-01-18:
Ah, well that code used to work before the v1.8.7 merge ;)  I guess I shouldn't be so naive to expect continuity across minor version numbers, especially since that functionality was probably just a stop-gap.  Oddly though, there is now no documentation at all for the `machine.I2C` [constructor](http://adafruit-micropython.readthedocs.io/en/latest/docs/library/machine.I2C.html#constructors).  Thanks for your recommendation, I will look into [bitbangio.I2C](https://circuitpython.readthedocs.io/en/latest/shared-bindings/bitbangio/I2C.html) - I don't think I need to worry about bus locking at this point, but I appreciate the warning.  Although, I am definitely interested in the new circuitpython functionality for the SAMD based Feathers.

tannewt, 2017-01-18:
The adafruit-micropython readthedocs is out of date since the name change. It is documented here: https://circuitpython.readthedocs.io/en/latest/docs/library/machine.I2C.html?highlight=machine.i2c

Also, the docs for `bitbangio` are more likely to stay in sync because they are defined [alongside](https://github.com/adafruit/circuitpython/blob/master/shared-bindings/bitbangio/I2C.c#L39) the interface instead of in a [separate file elsewhere](https://github.com/adafruit/circuitpython/blob/master/docs/library/machine.I2C.rst).