adafruit/circuitpython #84: m0: "'int' object is not callable" while setting the PWM duty cycle

deshipu, 2017-01-23:
```
>>> p = nativeio.PWMOut(board.D13)
>>> p.duty_cycle(50)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'int' object is not callable
```

tannewt, 2017-01-23:
Thats not the right way to use it. CircuitPython uses properties to represent state, not functions.

Try:
```
p = nativeio.PWMOut(board.D13)
p.duty_cycle = 50
```

I'll leave this open to add an example to the docs and also to switch to 16 bit duty_cycle to match `AnalogIn` and `AnalogOut`.

deshipu, 2017-01-23:
Aaah, I see. I didn't expect this, because that's a side effect. Attribute assignment shouldn't have side effects. You don't do `file.write = "hello"`, you call a function for doing input and output.

deshipu, 2017-01-23:
Otherwise you are going to have code like this:

```
>>> for p.duty_cycle in range(255):
...     time.sleep(0.01)
...     
```

tannewt, 2017-01-23:
I disagree. The duty_cycle is a state of the PWM output. It stays the same until its set to something else.

I agree file.write should be a function because its a modification of a larger piece of state (file.data if you will)

deshipu, 2017-01-23:
You also can't do `file[32] = byte` or `file[0:5] = "hello"`, even though those would be logical for the "larger piece of state".

tannewt, 2017-01-23:
Just because you can't do it, doesn't mean it doesn't make sense. Functions are for actions like sleep 1 second, blink ten times and turn left. Properties/attributes are for state which exists even when not actively used.

deshipu, 2017-01-23:
Do you think the example code I gave above is good and would you like to see it in the programs that you are going to have to read?

tannewt, 2017-01-23:
No, I don't think its good. However, I don't think the possibility of code like that is reason enough to use functions to represent state. Thats not what functions are. Its what variables, attributes and properties are. From [here](https://www.codecademy.com/courses/functions-in-javascript-2-0/0/1):

> A function is a block of reusable code. It is useful because you can execute it many times

deshipu, 2017-01-23:
Please don't quote kindergarten explanations about how a function could be visualized by beginning programmers. We both know what a function is and what is involved in its execution, down to the individual machine instructions. And we both know that the properties are no different technically -- if you really wanted to, you could replace all the function calls in the whole CircuitPython with property assignments (using tuples when you need multiple arguments). The difference is purely on the user side, in the appearances, meaning and, pardon my jargon, affordances (I really don't know a non-jargon term for this, sorry).

One of common design practices for programming languages (and their libraries) is making the looks of a particular statement reflect, roughly, the cost of executing it and any side effects it might have (see "Advanced Programming Language Design" by Finkel). The idea is that common, fast, simple, internal operations should be performed using operators; less common, slower; complex operations with side effects should be done with functions and statements, and the rare, slow, complicated operations may require several statements. Most modern programming languages follow this pattern more or less intentionally, and that's why fast programs tend to look light, and slow programs tend to be heavier on the eyes. This also gives the programmer a great heuristic about how well they are doing in keeping their programs lean. That is probably why we like properties so much -- they *feel* lighter and simpler, because all those years learning programming we've been conditioned to prefer them over the more expensive and thus more verbose operations.

However, we must not betray this expectation on which so many programmers unconsciously rely. We must not put expensive, slow, complex operations with side effects into operators -- including the assignment operator. If you do that, people will start writing programs that on the first sight feel like they should be faster and better, but that are actually not -- they will have been tricked, cheated. My code example above is just something that I immediately came up as soon as I thought about optimizing my code.

There is a number of expectations about the assignment operator that people have -- that it's idempotent , instant and safe. By using it to communicate with a hardware peripheral you are betraying their expectations -- because that particular operation is neither idempotent (it resets the timer), instant (writing to the timer registers takes more time than to a normal variable) nor safe (it has a side effect, possibly frying a LED or a motor that can't take a particularly large duty cycle). Failing expectations leads to surprises, and surprises lead to mistakes and frustration.

I'm well aware that you have already decided about this, and I'm not going to change your mind in this particular case. I'm writing this tirade so that better decisions could be made in the future. For this particular bug, an example in the `PWMOut` documentation would have been enough to fix my confusion (but not to prevent it in the first place). Sorry for ranting.

willingc, 2017-01-23:
Let's try to keep things civil and productive... 👍 

@tannewt's example shows the correct Python syntax for setting a property of a Python object.

> p = nativeio.PWMOut(board.D13)
p.duty_cycle = 50

Since my understanding is that CircuitPython will have a strong educational "learning Python" emphasis, it's likely better to err on the side of being Pythonic.

@deshipu While I see merits of some of your points when developing for high-performance with experienced engineers, it's likely best for learners and younger student experimenters to see a Python interface that reflects what they would learn in other Python books and resources.

willingc, 2017-01-23:
+1 for documentation in the docs and the source code.

tannewt, 2017-01-23:
Thanks for the feedback! I think this is a good discussion to have. 

One reason I like using properties for state (even though there is underlying bus logic) is that it does make it feel immediate and accessible. Learning the fundamental concept of state versus function can be really tricky because, as you point out, the line is very blurry. Treating a full circuit as local and immediate can help connect the dot between state in python and whether an led is lit or not.

commit 9f6da7b2 (2017-02-01): Add more PWMOut examples and rename the duty constructor argument
to duty_cycle so that its consistent with the attribute.

Fixes #84
closed 2017-02-01 by tannewt