adafruit/circuitpython #91: Configurable clock stretching

deshipu, 2017-01-28:
It would be nice to have this merged: https://github.com/micropython/micropython/pull/2550

Some sensors, like the BNO055, require more time than the default.

tannewt, 2017-02-13:
Is this only needed for `bitbangio.I2C`?

deshipu, 2017-02-13:
Well, the Feather M0 doesn't have `bitbangio`.

tannewt, 2017-02-13:
Ok, I was confused by your reference to the pull request. Can you please be more specific what methods need a longer timeout? Is it both reading and writing on `nativeio.I2C`?

deshipu, 2017-02-13:
I think this is a little bogus. I'm investigating the issue with the BNO055 right now, and it doesn't seem like the clock stretching timeout is actually the problem -- I thought so at the beginning, because I saw this:
https://github.com/adafruit/Adafruit_BNO055/blob/master/Adafruit_BNO055.cpp#L60-L63

However, it seems that this only happens when certain commands are sent quickly one after the other, and can be worked around by adding delays between them. I'm now experimenting with this.

tannewt, 2017-02-13:
Ok, keep me posted. I thought it might also be this timeout which is just cycles: https://github.com/adafruit/circuitpython/blob/master/atmel-samd/common-hal/nativeio/I2C.c#L77

deshipu, 2017-02-13:
I got the BNO055 driver to work at the default speed without any additional tuning by just adding some delays around the mode setting commands. We no longer *require* this feature for this particular sensor.

However, it may still prove useful in the future, and it may be useful when debugging similar issues with the drivers, so I wonder if we want this patch ported over anyways.

tannewt, 2017-02-14:
Could you explain what the patch does? I'm confused because it doesn't seem to change code that we share with MicroPython.

deshipu, 2017-02-14:
A slave device can hold the clock line down at any point during the transmission to signal to the master that its buffers are full and it needs time to process them -- the master has to check the clock line every time it releases it, and make sure it goes high. If it doesn't, it means that the slave is requesting a pause, and the master has to wait for the clock line to go high before doing anything else. That's called clock stretching.

However, for practical purposes, it would be bad for the master to wait indefinitely when the clock line is low -- because that is usually a sign of something bad happening and the bus getting stuck. That's why usually there is a timeout for that -- if the clock line is held low for longer than this timeout, the master will give up and just raise an error.

The patch referenced in this issue lets you configure this timeout, so that you can adjust it in case you have some devices on your bus that tend to take a long time processing their data, and may stretch the clock beyond the default of 255µs.

tannewt, 2017-02-15:
Yeah, I get what it is. I just don't know exactly what needs to change in CircuitPython. For now, I'm closing this issue pending a real world example. What we need to change will be clear when we have that and then it can be reopened. I don't want to leave it open because I'd like all open issues to be actionable so they can be closed for good and leave us with no open issues.

closed 2017-02-15 by tannewt

deshipu, 2017-02-15:
The change is quite simple. You need to have the stretch limit configurable here:

https://github.com/adafruit/circuitpython/blob/master/shared-module/bitbangio/I2C.c#L52

and also to raise an error when it is exceeded.

deshipu, 2017-02-15:
The changes in the linked pull request should apply directly to that file.

tannewt, 2017-02-15:
What about nativeio? Shouldn't they both be updated? These questions are
why I'd rather wait until we actually have a use case.

deshipu, 2017-02-15:
Here's a use case you can use: the SI7021 sensor that I wrote a driver for recently can make measurements in two modes. One, used in my driver, makes it become unresponsive to i2c messages while the measurement is happening, so you can keep pinging it to see when it's done. The other makes the sensor pull down the clock line (doing clock stretching) until the measurement is complete. Since a single conversion can take up to 12ms, and that's probably more than any default timeout, we can use that sensor for testing this feature.