adafruit/circuitpython #92: Impossible to extend descriptors

deshipu, 2017-01-28:
If I extend a descriptor (such as one of the registers from `adafruit_register`) using `super()`, it no longer has the ability to refer to itself through the `self` variable.

For example, with this code:

```
class _ScaledReadOnlyStruct(Struct):
    def __init__(self, register_address, struct_format, scale):
        super().__init__(register_address, struct_format)
        self.scale = scale

    def __get__(self, obj, objtype=None):
        result = super().__get__(obj, objtype)
        return tuple(self.scale * v for v in result)

    def __set__(self, obj, value):
        raise NotImplementedError()
```

I am getting this error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "bno055.py", line 34, in __get__
  File "adafruit_register/i2c_struct.py", line 43, in __get__
  File "adafruit_register/i2c_struct.py", line 42, in __get__
AttributeError: '' object has no attribute 'buffer'
```

The offending line is:

```
            obj.i2c_device.writeto(self.buffer, end=1, stop=False)
```

Note, that passing `self` to `super()` explicitly, (a la Python 2) makes it work:

```
class _ScaledReadOnlyStruct(Struct):
    def __init__(self, register_address, struct_format, scale):
        super(_ScaledReadOnlyStruct, self).__init__(
            register_address, struct_format)
        self.scale = scale

    def __get__(self, obj, objtype=None):
        result = super(_ScaledReadOnlyStruct, self).__get__(obj, objtype)
        return tuple(self.scale * v for v in result)

    def __set__(self, obj, value):
        raise NotImplementedError()
```

tannewt, 2017-02-20:
I tried to trigger this in a test but can't reproduce it. Any ideas? Heres the test:

```
class Base:

    def __init__(self):
        self.a = 1

    def __get__(self, obj, objtype=None):
        print("in Base meth", self.a)
        return self.a

class Sub(Base):

    def __get__(self, obj, objtype=None):
        print("in Sub meth")
        return super().__get__(obj, objtype)

class Outer:
    b = Sub()

a = Outer()
a.b
```

tannewt, 2017-06-18:
@deshipu Is this still an issue? If so, please reopen.

closed 2017-06-18 by tannewt