adafruit/circuitpython #95: I2C bus init error when specifying frequency on M0

deshipu, 2017-02-11:
On Feather M0, when I try to specify the clock frequency for the I2C bus, I get an OSError:

```
>>> i2c = I2C(SCL, SDA, frequency=10000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: I2C bus init error
```

In additional, after the creation of that I2C object failed, any subsequent attempts at creating it result in a "Pin in use" errror:

```
>>> i2c = I2C(SCL, SDA)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin PA23 in use
```

deshipu, 2017-02-11:
That means that I can't test the BNO055 sensor on the Feather M0.

tannewt, 2017-02-12:
Ok, sorry about this. It'll be a couple more days until I can fix it.

tannewt, 2017-02-13:
Why are you trying to run the bus at 10kHz? The datasheet says it can run up to 400kHz. 100kHz should work fine too.

deshipu, 2017-02-13:
Yeah, those are too fast for this sensor, at least with the default clock stretching timeout.

tannewt, 2017-02-13:
Ah, ok. It sounds like this really is that we need to lengthen the clock stretch timeout for `nativeio.I2C`. Can we consolidate this with #91? 10kHz seems non-standard enough that its ok not to support it.

deshipu, 2017-02-13:
If that value is invalid, it should raise a `ValueError` and exit cleanly, cleaning up all the pins afterwards.

tannewt, 2017-02-13:
Ah true. Will do that.

closed 2017-02-13 by tannewt

deryakitis, 2019-08-07:
Hi, when I try to run the circuitpython code below for an i2c of vcnl sensor and crickit robot on feather m0 express it gives me an ''SCL in use" error. Do you have an idea about how can I fix that? 

import time
import adafruit_vcnl4010
import board
import busio
from adafruit_crickit import crickit

i2c = busio.I2C(board.SCL, board.SDA)
while not i2c.try_lock():
    pass
print("I2C addresses found:", [hex(device_address)
                               for device_address in i2c.scan()])
i2c.unlock()
sensor = adafruit_vcnl4010.VCNL4010(i2c)
motor_1 = crickit.dc_motor_1
motor_2 = crickit.dc_motor_2
while True:
    motor_1.throttle = 1  # full speed forward
    motor_2.throttle = -1 # full speed backward
    time.sleep(1)

    motor_1.throttle = 0.5  # half speed forward
    motor_2.throttle = -0.5 # half speed backward
    time.sleep(1)

    motor_1.throttle = 0  # stopped
    motor_2.throttle = 0  # also stopped

    time.sleep(1)

    motor_1.throttle = -0.5  # half speed backward
    motor_2.throttle = 0.5   # half speed forward
    time.sleep(1)

    motor_1.throttle = -1  # full speed backward
    motor_2.throttle = 1   # full speed forward
    time.sleep(1)

    motor_1.throttle = 0  # stopped
    motor_2.throttle = 0  # also stopped
    time.sleep(0.5)
  
    proximity = sensor.proximity  # Proximity has no units and is a 16-bit
                                  # value.  The LOWER the value the further
                                  # an object from the sensor (up to ~200mm).
    print('Proximity: {0}'.format(proximity))
    ambient_lux = sensor.ambient_lux
    print('Ambient light: {0} lux'.format(ambient_lux))
    time.sleep(1.0)

and the error is:
code.py output:
Traceback (most recent call last):
  File "code.py", line 8, in <module>
ValueError: SCL in use

line 8 is:
i2c = busio.I2C(board.SCL, board.SDA)

caternuson, 2019-08-07:
When you import the crickit library, it sets up the I2C bus:
<https://github.com/adafruit/Adafruit_CircuitPython_Crickit/blob/master/adafruit_crickit.py#L382>
So when you try to do it again with:
```python
 i2c = busio.I2C(board.SCL, board.SDA)
```
It hits that error since the pins, both `SCL` and `SDA`, are in use. It only mentions `SCL` since that's the first one it tries to use.

Also, you don't need to worry about doing the lock and the scan. So change all this:
```python
i2c = busio.I2C(board.SCL, board.SDA)
while not i2c.try_lock():
pass
print("I2C addresses found:", [hex(device_address)
for device_address in i2c.scan()])
i2c.unlock()
sensor = adafruit_vcnl4010.VCNL4010(i2c)
```
to just this:
```python
sensor = adafruit_vcnl4010.VCNL4010(board.I2C())
```

deryakitis, 2019-08-07:
Firstly, thaks for the quick response!

I inverted the code into the way you said, now I am getting the same error
but for SDA.

Error:

Traceback (most recent call last):
  File "code.py", line 7, in <module>
ValueError: SDA in use

Code:

import time
import adafruit_vcnl4010
import board
import busio
from adafruit_crickit import crickit

sensor = adafruit_vcnl4010.VCNL4010(board.I2C())
#i2c = busio.I2C(board.SCL, board.SDA)
#while not i2c.try_lock():
#   pass
#i2c.unlock()
#sensor = adafruit_vcnl4010.VCNL4010(i2c)

Carter Nelson <notifications@github.com>, 7 Ağu 2019 Çar, 12:15 tarihinde
şunu yazdı:

dhalbert, 2019-08-07:
There's an underlying problem that the Crickit library doesn't use `board.I2C()`. It creates its own `I2C` object: 

https://github.com/adafruit/Adafruit_CircuitPython_Crickit/blob/617bb0787f2c61283d248632a62b27be80f64b29/adafruit_crickit.py#L382

That should be fixed. In the meantime you can do this:

```
i2c = crickit.seesaw.i2c_device.i2c
```
and then use that `i2c` object for the VCNL.

caternuson, 2019-08-07:
oh, yah. good catch. thanks @dhalbert 

@deryakitis here you go:
```python
Adafruit CircuitPython 4.0.2 on 2019-06-27; Adafruit Feather M4 Express with samd51j19
>>> import time
>>> import adafruit_vcnl4010
>>> import board
>>> import busio
>>> from adafruit_crickit import crickit
>>> sensor = adafruit_vcnl4010.VCNL4010(crickit.seesaw.i2c_device.i2c)
>>> sensor.ambient_lux
296.0
>>>
```

commit 9ed3e11a (2017-02-13): atmel-samd: Clean up I2C correctly when an initialization error occurs. Fixes #95.