adafruit/circuitpython #96: atmel-samd: reduce function not defined/enabled

tdicola, 2017-02-17:
Noticed the python built-in reduce function isn't defined on the samd port.  Very likely there's an obscure mpconfigport flag to track down and flip on.  Example using it and error:
```
>>> reduce(lambda x, y: x+y, [1, 2, 3, 4], 0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'reduce' is not defined
```
This would be handy to track down and flip on if space allows since functional style programming is a nice highlight of Micro/CircuitPython.  As a workaround though you can just iterate through collections and accumulate values manually with a for loop.

tannewt, 2017-02-19:
`reduce` was removed as a built-in from python3 so this is working as intended. Its recommended to use a for loop instead. https://docs.python.org/3.0/whatsnew/3.0.html#builtins

closed 2017-02-19 by tannewt

tdicola, 2017-02-20:
Oh haha wow didn't even realize they dropped it there.  I guess they moved it to a functools module that someone could implement later.