adafruit/circuitpython #97: Huzzah M0 wifi

fhelje, 2017-02-19:
I was wondering if I can get wifi working with the Adafruit Huzzah M0 wifi and circuitpython? It would be awesome if that worked. Since the dev experience is nicer with the SAMD board.

tannewt, 2017-02-19:
What do you mean by Huzzah M0? Do you mean this https://www.adafruit.com/product/3061 ?

What part of the dev experience are you looking to improve? The usb mass storage?

fhelje, 2017-02-20:
Thanks for the god work so far. Like the changes you are doing. They make a lot of sense at least for me that is new to this hardware stuff.

Yes usb mass storage shortens the feedback when developing. When I use my M0 proto board I can just develop of the usb but with the esp8266 I need to upload the files to the board. I could probably shorten the feedback loop by setting up the webrepl but that has been a bit sketchy. So I need to shutdown the terminal I have connected upload the file an reconnect.

But what I'm missing on the M0 is wifi. I know there is a M0 board with wifi but I haven't been able to figure out if I can use wifi with circuitpython on that board. If it already works it would be great if documentation is updated to reflect this. I far as I could figure out it was not supported.

tannewt, 2017-02-20:
Thanks! Yeah, there is a M0 + Wifi feather but its not our focus for CircuitPython. We suspect the M0 doesn't have enough resources to do both CircuitPython and Wifi at the same time. The ESP8266 already does this though.

For the ESP8266, ampy has a run command that makes it easy to upload and run a script. Right now it doesn't work well if the script doesn't stop but we could add CTRL-C support there. That way you'd just have to call run again. Would that work instead?

fhelje, 2017-02-23:
Yes that would probably work.

closed 2017-02-23 by fhelje