adafruit/circuitpython #98: No reboot after "import webrepl_setup"

fede2cr, 2017-02-20:
Running CP version 0.8.4. While trying to configure WebREPL, the boards offers to reboot and gets halted. A workaround is pressing the hard reset button on the ESP8266.

Details:
On a fresh ESP8266 huzzah feather, I import webrepl_setup, and after following the wizard, it asks me if I want to reboot the board.
If I say yes, I get inside the REPL:
"To enable WebREPL, you must set password for it
New password: greencore
Confirm password: greencore
Changes will be activated after reboot
Would you like to reboot now? (y/n) y
state: 5 -> 0 (0)
rm 0
del if0
bcn 0
del if1
usl
>>> 
 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset"

The boards just stays like that with the onboard red LED ON permanently.
Everything works perfect if I press the RESET button on the ESP8266.

Note: This feather actually has the modules adafruit_pca9685, adafruit_register and adafruit_bus_device, and is connected to a stacked motor featherwing.

Please let me know if I can provide any other details to replicate.

tannewt, 2017-02-21:
It looks like this simply calls into the ESP SDK so theres no bug in CircuitPython. I'll add a note to the setup that suggests manually resetting if it hangs.

tannewt, 2017-02-21:
Done in 44e5d360

closed 2017-02-21 by tannewt

fede2cr, 2017-02-21:
Thanks for your help, good solution.

tannewt, 2017-02-21:
Glad you like it! Let me know if theres anything else I can help with.