adafruit/circuitpython #101: Switch nativeio.I2C write to DMA.

tannewt, 2017-02-21:


tannewt, 2018-05-24:
I2C is slow enough that this isn't useful.

closed 2018-05-24 by tannewt

ladyada, 2018-05-24:
is there an issue for SPI DMA? (i didnt see) *that* would be useful, especially for TFTs

tannewt, 2018-05-24:
SPI is already DMAed. (QSPI isn't currently though.)

ladyada, 2018-05-24:
oooh do we wait for DMA to complete before returning from a transmit?

tannewt, 2018-05-24:
Yup, we still block. The DMA makes the transfer immune to interrupts and things.

ladyada, 2018-05-24:
ok i think in framebuf maybe that's where i'd optimize it so it does not block. all good!

tannewt, 2018-05-24:
Yup agreed. We should revamp display support and include this.

deshipu, 2018-05-24:
The problem with non-blocking SPI transfers is that we handle CS toggling in software, so we have to wait for the transfer to finish anyways, because toggling CS before it finishes would break things. Hardware CS could be one possible solution to that.

ladyada, 2018-05-24:
oh true... hmm, we could do that as part of the SPI locking process - like, CS is toggled by the DMA interrupt/callback, but you cant use it till then. a little like arduino transactions. just thinking outloud