adafruit/circuitpython #106: WebREPL fails to load after a softreset

fede2cr, 2017-02-22:
I have a freshly flashed ESP8266 with CP 0.9.0 testing the WebREPL code.
I've ran "import webrepl_setup" and configure it properly.
I upload to it a boot.py, without comments on webrepl to activate at boot:
```python
import webrepl
webrepl.start()
```
If I perform a hard reset, WebREPL comes up properly. **But** if instead I do a soft-reset with ``CTRL+D`` it prints:
```
WebREPL is not configured, run 'import webrepl_setup'
```
After this, if I do a hard reset, WebREPL comes back up again properly.

Please let me know if I can be of any help or you need me to runs test, etc.
Thanks in advance

tannewt, 2017-02-23:
Thanks for the report @fede2cr. This is a MicroPython bug that Tony actually found in December. I don't know much about webrepl so I can't fix it myself.

Here is the link to the upstream bug: https://github.com/micropython/micropython/issues/2659 Please follow up there. Thanks!

closed 2017-02-23 by tannewt

fede2cr, 2017-02-24:
Thanks, will do.