adafruit/circuitpython #108: SPI write doesn't support start and end keyword arguments.

tdicola, 2017-02-24:
This is odd, from the code it looks like the nativeio.SPI class write function should support the start and end keyword parameters to support writing a slice of the specified buffer.  However when I try this I get an error "TypeError: function does not take keyword arguments."  Probably a bit of macro magic or other config missing for the function?  

Here's an example:
```
>>> import board
>>> import nativeio
>>> import adafruit_bus_device.spi_device
>>> spi = nativeio.SPI(board.SCK, board.MOSI, board.MISO)
>>> cs = nativeio.DigitalInOut(board.D6)
>>> spi_device = adafruit_bus_device.spi_device.SPIDevice(spi, cs)
>>> with spi_device as locked_spi_device:
...     locked_spi_device.write(bytearray([0xFF, 0xFF, 0xFF]), start=0, end=1)
... 
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "<stdin>", line 2, in <module>
TypeError: function does not take keyword arguments
>>> 
```

tannewt, 2017-02-24:
Yup, sounds like a macro problem. I'll look today.

tannewt, 2017-02-25:
Sorry, got distracted trying to win bytes back. I haven't forgotten about this.

tdicola, 2017-02-27:
Oh no worries it's not blocking anything for me.  Good to check out and fix for next release.

tannewt, 2017-02-27:
What version are you running? It works for me on:
```
Adafruit CircuitPython 0.9.0-12-g3fd19c60-dirty on 2017-02-25; Adafruit Gemma M0 with samd21e18
>>> import nativeio
>>> import board
>>> s = nativeio.SPI(pin.PA09, MOSI=pin.PA07, MISO=pin.PA06)
>>> s.try_lock()
<SPI>
>>> s.write(bytearray([0xFF, 0xFF, 0xFF]), start=0, end=1)
```

tannewt, 2017-02-28:
I believe Tony was on 0.8.4 so this isn't a bug.

closed 2017-02-28 by tannewt