adafruit/circuitpython #110: atmel-samd: Big integer support

tdicola, 2017-03-07:
Filing an issue to remember, let's investigate big integer (> 31 bit values) support for the samd21 port.  In some discussion we think it's turned off right now and would fail.  Let's see what the impact would be to turn on and support.  It's not a super common case but core python to support arbitrary size integers.  Won't be as fast as the optimized 31-bit packed int values, but would let people port desktop python code over with less surprises when numbers get big.

willingc, 2017-03-24:
Yep. Just ran into this when going through the [8266 REPL tutorial}(https://adafruit-micropython.readthedocs.io/en/latest/docs/esp8266/tutorial/repl.html#using-the-repl) with the samd.
```
>>> 12**34
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: small int overflow
>>>
```

tannewt, 2017-03-24:
FYI, @willingc the latest docs are here:
https://circuitpython.readthedocs.io/en/latest/index.html The tutorial
still has that issue but the API docs will be more up to date.

I need to clean that other tutorial stuff up. My intention is for
ReadTheDocs to be mostly reference and use the Adafruit Learning System for
tutorials and guides.

tannewt, 2017-03-28:
Does anyone know of any specific use cases where we need this? OneWire 64 bit ids is one place they could be handy.

The MPZ implementation adds ~5000 bytes to the Feather M0 express build. The longlong implementation is ~3000 bytes. We have ~80 bytes left on the Arduino Zero build. I'd hate to split functionality between express/non-express so starkly.

tannewt, 2017-09-01:
I'll turn this on for the SAMD51 boards with #178.

tannewt, 2017-10-09:
Looks like enabling a LONGINT impl will break our mpys. Maybe we can change this so they are compatible?

dhalbert, 2017-10-10:
>Looks like enabling a LONGINT impl will break our mpys. Maybe we can change this so they are compatible?

After looking at this, I think this is not going to be an issue. `mpy-cross` is already set to use `MICROPY_LONGINT_IMPL_MPZ` for all the .mpy's it produces, so that will not change by turning on longints in our port. `tools/mpy-tool.py` cares about whether the implementation supports longints or not, but it's used to build internal frozen modules, not `.mpy` files.

tannewt, 2017-10-10:
OK perfect! That should make it easy.

commit 9db321d5 (2017-11-01): atmel-samd: Enable longints for M4 builds.

Fixes #110
closed 2017-11-01 by dhalbert