adafruit/circuitpython #113: Always allow variable frequency in PWMOut?

tdicola, 2017-03-24:
I noticed PWM out has a concept of allowing frequency changes, but it has to be explicitly enabled with variable_frequency=True in the initializer.  Any reason to not just always default this to on for more flexibility?  Curious if there's any code or hardware impact.

tannewt, 2017-03-24:
We could default it to on. When it is on it causes the underlying code to "claim" a whole timer rather than being willing to share. When the frequency is fixed multiple PWMOuts can share a timer for the same output frequency (like motors and servos). Its a trade-off between frequency flexibility and pin flexibility. Let me know if you still think its worth changing. Its really a matter of which case we think is more likely, flexible frequency or multiple outputs of the same frequency.

tannewt, 2017-03-27:
I'm going to leave it as is. Please reopen if you disagree.

closed 2017-03-27 by tannewt