adafruit/circuitpython #114: usb_hid: "USB Busy" errors trying to use the mouse device

dhalbert, 2017-04-09:
I added mouse support to `adafruit_hid`, but when I tested the code I got `OSError: USB Busy` errors in the REPL when I tried to send a report.

I found a device on usage page 1 with usage 2. I'm not sure what the USB report should be. I looked through the C++ code and it was hard to find the right descriptor or even interpret the ones I found. I was assuming it was four bytes: one for  buttons, and three for x, y, and wheel. But that may be wrong. Could you point me to the right place to look for it?

Is the `USB Busy` error due to the mouse conflicting with the keyboard device? I did some websearching and it appears that one might want to create a a "multiple Top-Level Collection (TLC)" HID device". But I haven't studied this in detail.

I have an important use case for instantiating the mouse and keyboard devices simultaneously, because I want to be able to simulate "Shift-leftclick", etc.

Here's the error and the code.
```
>>> from adafruit_hid import Mouse
>>> m = Mouse()
>>> m.move(0,0,0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_hid/mouse.py", line 83, in move
OSError: USB Busy
>>>
```

```
import usb_hid

class Mouse:
    """Send USB HID mouse reports."""

    LEFT_BUTTON = 1
    RIGHT_BUTTON = 2
    MIDDLE_BUTTON = 4
    ALL_BUTTONS = LEFT_BUTTON | RIGHT_BUTTON | MIDDLE_BUTTON

    def __init__(self):
        """Create a Mouse object that will send USB mouse HID reports."""
        self.hid_mouse = None
        for device in usb_hid.devices:
            if device.usage_page is 0x1 and device.usage is 0x02:
                self.hid_mouse = device
                break
        if not self.hid_mouse:
            raise IOError("Could not find an HID mouse device.")

        # Reuse this bytearray to send mouse reports.
        # report[0] buttons pressed (LEFT, MIDDLE, RIGHT)
        # report[1] x movement
        # report[2] y movement
        # report[3] wheel movement
        self.report = bytearray(4)

    def press(self, buttons):
        "Press the given mouse buttons."
        self.report[0] |= buttons
        self.move(0, 0, 0)

    def release(self, buttons):
        "Release the given mouse buttons."
        self.report[0] &= ~buttons
        self.move(0, 0, 0)

    def release_all(self):
        "Release all the mouse buttons."
        self.report[0] = 0
        self.move(0, 0, 0)

    def click(self, buttons):
        "Press and release the given mouse buttons."
        self.press(buttons)
        self.release(buttons)

    def move(self, x_distance, y_distance, wheel_turn):
        "Move the mouse and turn the wheel as directed. Any or all arguments may be zero."
        self.report[1] = x_distance
        self.report[2] = y_distance
        self.report[3] = wheel_turn
        self.hid_mouse.send_report(self.report)
```

tannewt, 2017-04-09:
Sorry about this. I haven't actually tested the mouse part.

The error is coming from: https://github.com/adafruit/circuitpython/blob/master/atmel-samd/common-hal/usb_hid/Device.c#L82

My understanding is that this would occur when the host computer hasn't polled a previous report.

The USB descriptor is instantiated by ASF because of this: https://github.com/adafruit/circuitpython/blob/master/atmel-samd/boards/feather_m0_express/conf_usb.h#L111

It looks like the HID descriptor is here: https://github.com/adafruit/circuitpython/blob/master/atmel-samd/asf/common/services/usb/class/hid/device/mouse/udi_hid_mouse.c#L122 It says the report length is 8 although when it sends it is 4. Perhaps ASF doesn't work.

I can take a look this week. I've got taxes to do today. :-)

dhalbert, 2017-04-09:
I made the report 8 bytes, and got an error saying don't do that (probably you knew that would happen):
```ValueError: Buffer incorrect size. Should be 4 bytes```

I am close to done on taxes. Collecting the info is most of the work.

dhalbert, 2017-04-09:
Here's some more info. When I plug in a CircuitPython Feather M0, Windows detects two new USB HID devices. One is the keyboard, but the other one has an error and is not identified. In the first screenshot you can see the yellow exclamation-mark triangle on the device with the error. There are two keyboards listed: one is the M0. But there's only one mouse listed. See the error reported in the Device Status of the second screenshot: "**A top level collection was declared without a usage or with more than one usage.**" So it sounds like there maybe something wrong with the HID descriptor for the mouse.

[image]
[image]

dhalbert, 2017-04-14:
I am making some progress here. I took a USB traces of the Feather M0 being enumerated. I can see where the USB descriptors for the keyboard and mouse and being sent to the host. The keyboard descriptor is fine, but the mouse descriptor is being mangled in a strange way. Its bytes are being copied with interleaved zero bytes, and every other pair of bytes in the descriptor is being omitted.

Here is the descriptor, from https://github.com/adafruit/circuitpython/blob/master/atmel-samd/asf/common/services/usb/class/hid/device/mouse/udi_hid_mouse.c#L122, as you noted above:

```
//! HID report descriptor for standard HID mouse
UDC_DESC_STORAGE udi_hid_mouse_report_desc_t udi_hid_mouse_report_desc = {
	{
				0x05, 0x01,	/* Usage Page (Generic Desktop),       */
				0x09, 0x02,	/* Usage (Mouse),                      */
				0xA1, 0x01,	/*  Collection (Application),          */
				0x09, 0x01,	/*   Usage (Pointer),                  */
				0xA1, 0x00,	/*  Collection (Physical),             */
				0x05, 0x09,	/*     Usage Page (Buttons),           */
				0x19, 0x01,	/*     Usage Minimum (01),             */
				0x29, 0x03,	/*     Usage Maximum (03),             */
				0x15, 0x00,	/*     Logical Minimum (0),            */
				0x25, 0x01,	/*     Logical Maximum (1),            */
				0x75, 0x01,	/*     Report Size (1),                */
				0x95, 0x03,	/*     Report Count (3),               */
				0x81, 0x02,	/*     Input (Data, Variable, Absolute) */
				0x75, 0x05,	/*     Report Size (5),                */
				0x95, 0x01,	/*     Report Count (1),               */
				0x81, 0x01,	/*     Input (Constant),               */
				0x05, 0x01,	/*     Usage Page (Generic Desktop),   */
				0x09, 0x30,	/*     Usage (X),                      */
				0x09, 0x31,	/*     Usage (Y),                      */
				0x09, 0x38,	/*     Usage (Scroll),                 */
				0x15, 0x81,	/*     Logical Minimum (-127),         */
				0x25, 0x7F,	/*     Logical Maximum (127),          */
				0x75, 0x08,	/*     Report Size (8),                */
				0x95, 0x03,	/*     Report Count (3),               */
				0x81, 0x06,	/*     Input (Data, Variable, Relative) */
				0xC0,	/*  End Collection,                    */
				0xC0,	/* End Collection                      */
			}
};
```

and here are the payload bytes being sent to the host:
```
  0x05, 0x00, 0x01, 0x00, 
  0xa1, 0x00, 0x01, 0x00,
  0xa1, 0x00, 0x00, 0x00,
  0x19, 0x00, 0x01, 0x00,
  0x15, 0x00, 0x00, 0x00,
  0x75, 0x00, 0x01, 0x00,
  0x81, 0x00, 0x02, 0x00,
  0x95, 0x00, 0x01, 0x00,
  0x05, 0x00, 0x01, 0x00,
  0x09, 0x00, 0x31, 0x00,
  0x15, 0x00, 0x81, 0x00,
  0x75, 0x00, 0x08, 0x00,
  0x81, 0x00, 0x06, 0x00
```
Notice that if you drop every other byte in the payload (those bytes are all `0x00`), the payload lines match every other line in the original descriptor code: `0x05, 0x01, ..., 0xa1, 0x01, ... 0xa1, 0x01, ..., 0x19, 0x01`, etc.

This is weird but implies to me bytes are being copied as 16-bit words or something like that. There may be a misdeclaration somewhere where bytes should be words.

I will try to track down where this is being copied, but if you know where this might be happening, let me know. My guess is that the ASF code is probably OK, so it has something to do with the CP wrapper code or something in one of the numerous configuration `.h` files you had to set up.

dhalbert, 2017-04-14:
Side comment, you mentioned above:

> It looks like the HID descriptor is here: https://github.com/adafruit/circuitpython/blob/master/atmel-samd/asf/common/services/usb/class/hid/device/mouse/udi_hid_mouse.c#L122 It says the report length is 8 although when it sends it is 4. Perhaps ASF doesn't work.

If you are talking about here:
```
				0x09, 0x30,	/*     Usage (X),                      */
				0x09, 0x31,	/*     Usage (Y),                      */
				0x09, 0x38,	/*     Usage (Scroll),                 */
				0x15, 0x81,	/*     Logical Minimum (-127),         */
				0x25, 0x7F,	/*     Logical Maximum (127),          */
                                0x75, 0x08, /*     Report Size (8),                */
	                        0x95, 0x03, /*     Report Count (3),               */
```
in the above I think it's talking about 8 **bits** repeated 3 times for X, Y, and Scroll. I was confused by that at first too.

dhalbert, 2017-04-14:
I thought this perhaps might be a toolchain issue. But I looked in the .bin and I can easily find both the mouse and keyboard descriptors. So the mouse descriptor is intact in the object code.

tannewt, 2017-04-14:
Have you tried calling the ASF mouse helpers directly to see if they work? Thats where I started with the keyboard stuff.

dhalbert, 2017-04-14:
I'm not sure I know which helpers you mean. Do you mean like "press this mouse button", or the enumeration functions? Since the mouse device isn't enumerated properly, the host will not be expecting any reports from it.

So far I have looked at the kbd files and the corresponding mouse files in various places, since the kbd enumeration works fine. I haven't found any differences that seem significant.

Did you have to make changes in the `asf/` tree? I was thinking of downloading it and diff'ing the trees. Which version did you use?

I wonder if I should get an Arduino Zero or a J-Link for debugging.

tannewt, 2017-04-14:
Yeah, I mean the helpers that create the report.

I've made a few changes to ASF but nothing super large. Its version 3.32 IIRC.

Debugging is useful for hard faults but its hard to use the debugger on USB because of timing sensitivity. If you pause in the debugger you can hose the whole bus. Its not fun. J-Link + Metro M0 Express is my goto otherwise. Arduino Zero is much cheaper but doesn't have the external SPI flash.

PR #118 (2017-04-15): atmel-samd: HID report descriptor for mouse had to be COMPILER_WORD_ALIGNED
This fixes #114. When the USB HID descriptor for the mouse was sent to the host, it was garbled. I forced the descriptor to be word-aligned, and then it was sent back properly.

There are hints that descriptors should be word-aligned. See for instance https://github.com/adafruit/circuitpython/blob/master/atmel-samd/asf/common/services/usb/udc/udd.h#L312. In many but not all places in the ASF library, `USB_DESC_STORAGE` declarations are `COMPILER_WORD_ALIGNED`, but not all. Perhaps it's an accident that most of them wor [... cut]
closed 2017-04-15 by tannewt