adafruit/circuitpython #121: build fails for feather_m0_basic

g-ollivier, 2017-04-18:
Hello,

Build works for feather_m0_express but doesnt anemyore for feather_m0_basic
ld: region `FLASH' overflowed by 248 bytes

I checked with previous releases 0.9.3 and 0.9.4 and had the same error.

Best Regards
Guillaume

g-ollivier, 2017-04-18:
Re,
Which version of gcc-arm must be used now ? 
Thank you

g-ollivier, 2017-04-18:
Re,
Sorry for the noise.

Made it work with release 6-2017-q1 update
found here https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

Closing this issue.

closed 2017-04-18 by g-ollivier

dhalbert, 2017-04-18:
FYI: if on Ubuntu, `ppa:team-gcc-arm-embedded/ppa` seems to work fine and was easy to install:

```
sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
sudo apt update
sudo apt install gcc-arm-embedded python build-essential
```
I have used this both on 14.04 and 16.04.

tannewt, 2017-04-18:
No worries @g-ollivier. I always keep up with the latest. (The continuous build actually updates automatically.)

Thanks for the tips @dhalbert.