adafruit/circuitpython #122: Stale reference prevents reusing pins

ianrrees, 2017-04-21:
Perhaps this is expected behaviour (I just received my Feather M0 Express a little while ago, completely new to MicroPython), but this seems weird:

```
Adafruit CircuitPython 0.9.5 on 2017-04-14; Adafruit Feather M0 Express with samd21g18
>>> import digitalio
>>> import board
>>> led = digitalio.DigitalInOut(board.D13)
>>> led = "I'm not an LED anymore!"
>>> newVar = digitalio.DigitalInOut(board.D13)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Pin PA17 in use
```

tannewt, 2017-04-21:
Yeah, it is weird. Unfortunately Python usually makes no promises about when objects get deconstructed.

However! Python does have context managers that can make lifetimes clear and all of our hardware API uses it. So, when you want this to be clear you can:

```
import digitalio
import board
with digitalio.DigitalInOut(board.D13) as led:
  led.value = True
  led.value = False
led = "I'm not an LED"
with digitalio.DigitalInOut(board.D13) as led:
  led.value = True
  led.value = False
```

With the REPL its kinda clunky though. You'll need to do a soft-reset (CTRL-D) to reset the pin.

Thanks for the issue! I'm happy to hear people are using CircuitPython and would love to know the pain points.

closed 2017-04-21 by tannewt

ianrrees, 2017-04-21:
I don't know how MicroPython works internally, but if it does reference counting like cpython, then it should be possible to determine whether there are any references to an object that manages a particular hardware resource.  This way, even if the Python object hasn't been deconstructed (the refcount is 0, but the gc hasn't free()'ed the object), it might be possible to determine that the hardware resource can reused safely.

```
In [1]: import sys

In [2]: class Thing:
   ...:     pass
   ...: 

In [3]: persistent = Thing()

In [4]: sys.getrefcount(persistent)
Out[4]: 2

In [5]: led = persistent

In [6]: sys.getrefcount(persistent)
Out[6]: 3

In [7]: led = "not an led"

In [8]: sys.getrefcount(persistent)
Out[8]: 2
```

tannewt, 2017-04-21:
It doesn't do reference counting. The GC is currently a mark and sweep
algorithm.

The algorithm for determining whether a pin is in use or not is based on
the hardware registers themselves in case multiple objects use the same
hardware. So, we'd need a reliable call to the deinit method of the object.
(I forgot to mention you can call deinit() too.)

I don't think its too big of a deal but I'm happy to hear suggestions.

ianrrees, 2017-04-21:
Ahh, thanks for the quick and thorough feedback!  I agree that it's not really a big deal in practical use, but reported the bug because the error could be a bit mysterious to beginners.  Would it make sense to do something like:

```
if (hardware is in use) {
  garbage_collect();
  if (hardware is in use) {
    sad_face();
  }
}
```

tannewt, 2017-04-25:
Thats not a bad idea! I'll keep it in mind. Thanks!