adafruit/circuitpython #123: Problem with UF2 Bootloader on Windows 7

slzatz, 2017-05-03:
(Note if there is a forum or somewhere better to post this, please let me know.)

On Windows 7, using the M0 feather express, I see the Feather boot loader and can drag the circuitpython.uf2 file onto the bootloader drive and circuitpython will run fine.  However, I can't connect to the board serially.  So next step -- I load the adafruit driver m0 feather express driver, and then I can connect to the board but all I see when I hit control-c is 'usbusbusb...' and with the driver in place the circuitpython drive disappears.  Note that I can connect to the same board via linux box and connect serially with no problem and circuitpython is running fine and the circuitpython drive is also present.  

So the issue seems to be a Windows 7 issue and am wondering if there are any recommendations other than stop trying via Windows.

ladyada, 2017-05-03:
hiya just to get rid of one option, can you try updating the bootloader? the first bootloader we shipped had some weirdnesses on some win boxes. here's the procedure:
https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/uf2-bootloader#updating-the-bootloader

slzatz, 2017-05-03:
Thanks for the note and if this is ladyada, nice to meet you.  I'm the CEO of WebMD and we're located about 5 minutes from you on Hudson St.  I may be your biggest fan (and I am sure you have many) and am a loyal customer (electronics is a hobby of mine and I think your Feather series is brilliant).  In any event, I should have mentioned that the first thing I did was to update the bootloader according to those instructions when things didn't work and then I went through the whole sequence again.  It did not fix the problem.  The fact that I could take the same device and connect on a serial port on Linux seems to suggest that there may be something funky about my Windows environment that is interfering with the driver operating correctly.  In any event, open to suggestion as to what to do next although maybe I should try to find another Windows 7 machine and see if the same thing happens.

Steve

tannewt, 2017-05-03:
@slzatz Did you confirm the bootloader was updated?

Your problem is connecting when its running CircuitPython right? Not when its in bootloader me.

@dhalbert have you seen this too?

dhalbert, 2017-05-03:
@tannewt, I have only tried Windows 10, but I have one Windows 7 machine left I can try tonight.
 @slzatz , did you update with the v1.21.0 bootloader? There was an intermediate one that did not play well with the drivers.

A couple of related forum threads:
https://forums.adafruit.com/viewtopic.php?f=57&t=116294
https://forums.adafruit.com/viewtopic.php?f=57&t=115911 (see second page of this thread)

slzatz, 2017-05-03:
When home tonight I'll uninstall the driver, which allows me to see the drive and confirm I have 1.21.0.

slzatz, 2017-05-04:
The bootloader info is:

```
UF2 Bootloader v1.21.0 SFHR
Model: Feather M0
Board-ID: SAMD21G18A-Feather-v0
```
And it continues to only become visible when I delete the Adafruit Feather M0 Express Windows driver.  Happy to move any further updates I have to the forum if that's a better place for this discussion.

tannewt, 2017-05-04:
Weird! Does the serial work without the driver? I think here is fine for discussion.

The flakiness does sound like a clock issue but that new bootloader should be fixed. Do you have an oscilloscope by chance? I can get you a special build that will output the clock on a pin to check it.

dhalbert, 2017-05-04:
I tried this on a Windows 7 desktop machine (rather standard i3-2100 Dell). It had all the Adafruit drivers installed initially. CIRCUITPY showed up but FEATHERBOOT would not. Then I uninstalled the drivers one by one, and FEATHERBOOT appeared. I reinstalled the uninstalled drivers, and it still appeared. I rebooted, uninstalled, and reinstalled, and could not reproduce the missing FEATHERBOOT. So I can't get it to act up consistently.

dhalbert, 2017-05-04:
@slzatz Is it a USB 2.0 or 3.0 port? If you have a USB 2.0 hub interpose that and see if it makes any difference.

slzatz, 2017-05-04:
I am pretty sure it's USB 3 and I'll look for a USB 2 hub and let you know.

Without the driver, I don't see the board at all on any port.  So sequence is that I see the drive - both the feather drive in boot mode and the circuitpython drive once that's loaded but without the driver, I don't see the board on any port.  Load the driver and the board appears (usually something like COM16), the drive disappears and when I connect i get "USBUSB...." every time I hit CNTRL-C.  On the other hand, when I just walk over to a linux box with the board and plug it in I can see the drive and can connect serially via screen.

And sorry, no oscilloscope - could be my next Adafruit purchase.

slzatz, 2017-05-04:
Tried a USB 2 hub but no success -- with the driver installed the circuitpython disc drive does not appear and in this case while I can see the board on COM21, I can't connect at all through Putty.  Previously, i was getting "USB" printed to the console every time I hit cntrl-c but this time around could not connect at all.

willingc, 2017-05-04:
@slzatz What happens if you reboot the Windows 7 machine and try again?

slzatz, 2017-05-04:
Rebooted and with driver active, no feather/circuitpython drive appears.  At some point we may have to write it off to my Windows 7 machine and tomorrow I will have access to another and can see if that works or not.

tannewt, 2017-05-04:
Sorry for the trouble! What are the specs of the machine?

I'll be setting up Windows 7 on a spare computer I have here in the next
week or so and will test it out once I do.

slzatz, 2017-05-04:
No trouble at all.

Windows 7 Enterprise
Service Pack 1
i5 2.3 GHz
8 GB RAM
running on a 2015/6 Lenovo X1 Carbon

slzatz, 2017-05-06:
One more piece of information.  I didn't have access to another Windows 7 machine but when I plug the same board into a Windows 10 machine, it works as expected.  CIRCUITPY drive appears and I can connect serially through Putty.  So not sure what the deal is with my Windows 7 machine but same board works correctly under Windows 10.

willingc, 2017-05-06:
@slzatz Happy it works for you on Windows 10.

tannewt, 2017-08-01:
I'm gonna close this. I haven't heard any similar reports. Please reopen if its still an issue.

closed 2017-08-01 by tannewt