adafruit/circuitpython #124: "syntax errors" when using bitmapfont - sometimes

jerryneedell, 2017-05-04:
I have been experimenting with using bitmapfont to write text to an ILI9341 TFT display:
320x240 pixels

It works for the most part, but I often run into a response of:
"SyntaxError: invalid syntax"
when I try to write a new text string to the screen.

Here is an example of the sequence I am using:

`import busio
import digitalio
from board import *

from adafruit_rgb_display import color565
import adafruit_rgb_display.ili9341 as ili9341
import bitmapfont

# Configuratoin for CS and DC pins (these are FeatherWing defaults on ESP8266):
CS_PIN = D9
DC_PIN = D10
# Config for display baudrate (default is 32mhz, about as fast as the ESP supports):
BAUDRATE = 1000000

# Setup SPI bus using hardware SPI:
spi = busio.SPI(clock=SCK, MOSI=MOSI, MISO=MISO)

# Create the ILI9341 display:
display = ili9341.ILI9341(spi, cs=digitalio.DigitalInOut(CS_PIN),dc=digitalio.DigitalInOut(DC_PIN), baudrate=BAUDRATE)

display.fill(0)

bf= bitmapfont.BitmapFont(320,240,display.pixel)
bf.init()
bf.text('abc',0,0,color565(255,0,255))
bf.text('def',0,100,color565(255,0,255))

`
It usually works OK for "awhile" then I will get a SyntaxError when I try to add some text.

If I get the SyntaxError, sometimes I seem to be able to clear it by:
import gc
gc.collect()
then reissue the bf.text command but this is not completely reproducible.

I have also observed that when the SyntaxError occurs
gc.mem_free indicates around 2400 bytes free
then after the gc.collect() gc.mem_free() indicates about 6400 bytes free.

This is also not reliable - sometimes I get SyntaxErrro on the gc.  commands as well

It appears to be due to memory allocation issues, but it  the behavior is not consistent.

It may be my imagination, but I seem to run into this problem more if I execute the commands from REPL manually more than if I upload a .py script and execute it.

Sorry for the nebulous report. I'm trying got find a consistent behavior.

dhalbert, 2017-05-04:
In gitter, I mentioned that I sometimes see random syntax errors when using the REPL, complaining about an error on line 1 on stdin. I am not doing anything that fancy, and I am not using the frame buffer. When I see these, it seems as if the readline() processor might be doing something wrong.

@jerryneedell, are you copy/pasting (middle-click, right-click) these commands sometimes in the terminal window? I have seen apparently good copy/paste give syntax errors, as if it is picking up invisible characters and/or not actually pasting what I think it is pasting.

I think it would not be hard for debugging purposes to patch CPy to print out the offending line with the syntax error as a Python string. Then we could see what it's really reading: any invisible characters would be converted to a `\` escape sequence.

jerryneedell, 2017-05-04:
@dhalbert Good point about using copy/paste. I am often using either cut/paste or up-arrow when I get these Syntax Errors so they may well have been related to that.
I wil see if I get better results manually entering the commands. In support of this, I can't recall any cases when the syntax error occurred as during a script execution.  I think I have only seen them when using REPL. I will also pay much closer attention to this going forward.

dhalbert, 2017-05-04:
@jerryneedell Which terminal program on which OS are you using? I am seeing this on Windows with Tera Term, and may have seen it with Putty as well. It may not be related to the terminal program, but it could be a minor factor. readline() is expecting the terminal program to be emulating a VT100, and processes ESC sequences  for the arrow keys, etc.,

jerryneedell, 2017-05-04:
@dhalbert I am using "screen" via  linux (Raspian Jessie) on a Raspberry Pi  3 to connect to the M0 express.

jerryneedell, 2017-05-04:
an update: I used REPL to execute several writes as in the above example. I was able to execute several using readline features (up-arrow - edit line - execute) and eventually got a syntax error (always File "<stdin>" ,line 1 as noted by @dhalbert).  Once I received the error, I tried entering commands manually, but still received the same error. I was able to execute a display.fill(0) once, but not a second time??
I tried import gc and it worked, but any gc.mem_free() or gc.collect() attempts hals generated syntax errors.... I am able to execute help() but not help with any argument - that also generates a syntax error. 
I'm kind of stuck here so I'll do a control-d and move on.

jerryneedell, 2017-05-04:
It is clearly not just executing the commands that causes the problem - if I run a continuous loop from a script it works OK:
Here is an example that runs:

```
import time
import busio
import digitalio
from board import *

from adafruit_rgb_display import color565
import adafruit_rgb_display.ili9341 as ili9341
import bitmapfont

# Configuratoin for CS and DC pins (these are FeatherWing defaults on ESP8266):
CS_PIN = D9
DC_PIN = D10
# Config for display baudrate (default is 32mhz, about as fast as the ESP supports):
BAUDRATE = 1000000

# Setup SPI bus using hardware SPI:
spi = busio.SPI(clock=SCK, MOSI=MOSI, MISO=MISO)

# Create the ILI9341 display:
display = ili9341.ILI9341(spi, cs=digitalio.DigitalInOut(CS_PIN),dc=digitalio.DigitalInOut(DC_PIN), baudrate=BAUDRATE)

display.fill(0)

with bitmapfont.BitmapFont(320,240,display.pixel) as bf:
    try:
        while(True):
            display.fill(0)
            bf.text('abc',0,0,color565(255,0,255))
            bf.text('def',0,100,color565(255,0,255))
            bf.text('abc',100,0,color565(0,255,255))
            bf.text('def',200,200,color565(0,0,255))
            time.sleep(5)           
    except:
        pass
    finally:
        pass

```

FYI - if I stop - via control-c then do a gc.mem_free() I am down to 1600 bytes free.
gc.collect() restores it to 6888
But I did see one syntax error with gc.mem_free() but was able to re-enter it and run it.

The syntax errors are only via REPL. 

head scratching continues...

dhalbert, 2017-05-04:
It's beginning to come back to me that I think I could reliably generate syntax errors by doing the `gc` calls on the same line.
```
>>> import gc
>>> gc.collect(); gc.mem_free()
>>> gc.collect(); gc.mem_free()
```
I have a possibly faulty memory of having  to up-arrow and torepeat this command to get the error.

I don't have CPy to try right now; will try tonight.

jerryneedell, 2017-05-04:
@dhalbert I just rebooted and ran that sequence repeatedly with no errors on my M0 Express.
gc.mem_free() reveals > 15000 bytes free

dhalbert, 2017-05-05:
@jerryneedell  I built a version of 0.9.7 for M0 express that repeats the string you typed into the REPL and also writes out the characters again, with non-printing characters in hex. I have to zip it up to upload it here.

[repl-print-circuitpython-feather_m0_express-0.9.7.bin.zip](https://github.com/adafruit/circuitpython/files/977947/repl-print-circuitpython-feather_m0_express-0.9.7.bin.zip)

So try this out and see what's printed when you get syntax errors.

Example:

```
>>> print ("a space")
print ("a space")
 p r i n t 0x20 ( " a 0x20 s p a c e " )
a space
>>>
```

dhalbert, 2017-05-05:
@jerryneedell  Well, I just figured out how to generate consistent syntax errors. The REPL diagnostic I posted above shows that there's nothing wrong with the input. What I typed is what it's parsing. It has to do with order of operations. See the next comment...

dhalbert, 2017-05-05:
@jerryneedell @tannewt This sequence will consistently generate syntax errors. You start by pressing the reset button: a soft reset won't do it. No cutting and pasting is necessary. I just typed all this. Note that the `gc` operations are OK, and  `abc` generates `SyntaxError` the first time, and the correct error the second time. This is with the stock released firmware, not my own build.

I have also seen harder-to-duplicate states where it seems to get stuck in `SyntaxError` permanently.

_**Press the reset button, then type the input below at the prompts.**_
```
Auto-soft reset is on. Simply save files over USB to run them or enter REPL to disable.
Press any key to enter the REPL. Use CTRL-D to soft reset.

Adafruit CircuitPython 0.9.7 on 2017-05-01; Adafruit Feather M0 Express with samd21g18
>>>
>>> import gc
>>> gc.collect();gc.mem_free()
15728
>>> abc
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> abc
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'abc' is not defined
>>> 123
123
>>>
```

tannewt, 2017-05-05:
Awesome work @jerryneedell and @dhalbert! I'll take a look at it tomorrow if you don't fix it before then. :-)

jerryneedell, 2017-05-05:
@dhalbert Nice work! I think it's encouraging that it does not appear to be related to memory  shortage usage as I initially thought.  I'm also glad to hear you have seen it get stuck permanently as I have. Nice to have company ;-)

dhalbert, 2017-05-05:
@tannewt Similar error reported for MicroPython, but supposedly fixed some time ago:
https://github.com/micropython/micropython/issues/1787

tannewt, 2017-05-05:
Ok, I've reproduced it! Thanks @dhalbert.

```
Adafruit CircuitPython 0.9.7-7-g5ad42612 on 2017-05-05; Adafruit Metro M0 Express with samd21g18
>>> import gc
>>> gc.collect()
>>> 123
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> 123
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> 123
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> 123
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> abc
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```

tannewt, 2017-05-05:
As usual, @dhalbert was right on the trail. Turning on LTO moved the `line` variable from `pyexec_friendly_repl` higher in the stack than the dummy variable to measure the stack top. Moving to using `_estack` from the linker fixes the issue.

Here is a GC log of the repro case if you are curious: https://gist.github.com/tannewt/5a2ed9ff724d576b934a9393c5fe87ef

Fix will come shortly after I clean up the debug stuff I added.

jerryneedell, 2017-05-05:
Excellent! Great work @dhalbert and @tannewt - thanks for the quick turnaround one fixing this!

commit c6d539ac (2017-05-05): atmel-samd: Fix sporadic "syntax errors"

The GC was deleting memory that was in use because its scan of the
stack missed the very top. Switching to _estack fixes this by relying
on the location from the linker.

Fixes #124
closed 2017-05-05 by tannewt