adafruit/circuitpython #129: Lots of <3 on ESP8266 in the docs, but where are the tutorials for SAMD21 ??

tychop, 2017-05-13:
I want to do some tutorials but am always getting confronted with `import machine` which is not available on SAMD21....

For example... How do I control NeoPixels without import machine? Do I import digitalio, or pulseio?
I have two Adafruit M0 feathers which I want to use using circuitpython.

Am I not seeing/finding the tuts on this or are there none (yet)?

tannewt, 2017-05-13:
They are on learn.adafruit.com and some have been updated to include CircuitPython instructions. Libraries tend to have short examples as well.

Unfortunately the neopixel library doesn't: https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel

All you need to pass it is the pin from `board` and the length of the strip.

There are also these API docs for CircuitPython: https://circuitpython.readthedocs.io/en/latest/

tychop, 2017-05-13:
Yeah, I already found what you're suggesting and came up with this:

`import digitalio
import board
import neopixel

np = neopixel.NeoPixel(digitalio.DigitalInOut(board.D9), 8)
np[0] = (255, 0, 0)`

Unfortunatly, this doesn't work, so I am eager for example code that does work on a SAMD21 so I can figure out where I go wrong....

tannewt, 2017-05-13:
Try calling np.write() at the end. Feel free to ask questions here, on
Gitter or the adafruit forums.

tychop, 2017-05-14:
Got it !!!

The initialisation of the neopixel was wrong...
This does work:

`pin5 = board.D5`
`np = neopixel.NeoPixel(pin5, 8)`
`np[0] = (255, 000, 000)`

`np.write()`

closed 2017-05-14 by tychop

tannewt, 2017-05-14:
Ah right! I actually want to change it to the first form because composition (passing in a DigitalInOut) is better because then you could pass in some sort of other object with the same capabilities. For example, a GPIO expander library could provide something that was DigitalInOut-like.