adafruit/circuitpython #130: Problems building with internal frozen modules

dhalbert, 2017-05-15:
I have run out of RAM for Python code and need to incorporate some code into the firmware. I looked at the `atmel-samd/Makefile` and found I could do this:
```
$ make V=1 BOARD=feather_m0_express FROZEN_MPY_DIR=path/to/frozen
```
where `frozen` contains the `.py` files I want to build in. They get converted to `.mpy` and then `mpy-tool.py` merges and converts those to  `frozen_mpy.c`.

First problem: `mpy-tool.py` complained:
```
build-feather_m0_express/frozen_mpy.c:11:2: error: #error "incompatible MICROPY_LONGINT_IMPL"
```                     

That's because `mpy-tool.py` has a parameter `-mlongint-impl` which should be `none`, but it defaults to `mpz`. The Makefiles are not setting that up properly; it should be based on what's in `mpconfigport.h`. So for now I hacked the invocation of `$(MPY-TOOL)` in `py/mkrules.mk`:
```
-       $(Q)$(PYTHON) $(MPY_TOOL) -f -q $(BUILD)/genhdr/qstrdefs.preprocessed.h $(FROZEN_MPY_MPY_FILES) > $@
+       $(Q)$(PYTHON) $(MPY_TOOL) -mlongint-impl=none -f -q $(BUILD)/genhdr/qstrdefs.preprocessed.h $(FROZEN_MPY_MPY_FILES) > $@ 
```

Now `frozen_mpy.c` will compile, but I get LTO link errors later:
```
LINK build-feather_m0_express/firmware.elf
build-feather_m0_express/frozen_mpy.c:96:26: error: size of 'mp_qstr_const_pool' differ from the size of original declaration [-Werror=lto-type-mismatch]
 extern const qstr_pool_t mp_qstr_const_pool;
                          ^
../py/qstr.c:98:19: note: 'mp_qstr_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_const_pool = {
                   ^
../py/qstr.c:113:26: error: size of 'mp_qstr_frozen_const_pool' differ from the size of original declaration [-Werror=lto-type-mismatch]
 extern const qstr_pool_t MICROPY_QSTR_EXTRA_POOL;
                          ^
build-feather_m0_express/frozen_mpy.c:97:19: note: 'mp_qstr_frozen_const_pool' was previously declared here
 const qstr_pool_t mp_qstr_frozen_const_pool = {
                   ^
lto1: all warnings being treated as errors
lto-wrapper: fatal error: arm-none-eabi-gcc returned 1 exit status
compilation terminated.
```

So... this is a lot of detail. I could continue to pursue the errors, but thought you might have some suggestions. The frozen stuff may not have been tested since you started using LTO. Thanks.

tannewt, 2017-05-15:
The frozen module stuff isn't a high priority for me so I won't be able to spend any time dedicated to it.

However, I vaguely recall running into this with something else while enabling LTO. I think it had to do with structs of varying size. I didn't know it was possible but it has something to do with an array at the end of the struct. My solution if I remember right was to just fix the array size instead of leaving it flexible. I'm not sure thats possible in this case though.

I'd probably take the opposite approach of trying to reduce RAM usage. There are a couple ways I could look into for that:
* Increasing the heap size (just a statically allocated array) at the expense of the stack space.
* Forcing the SPI flash code to use the scratch sector rather than a sector in memory, which is 4k. This will slow it down though.
* Staring into the abyss of the heap tracking output and coming up with another idea. :-)

Sorry I don't have a silver bullet.

dhalbert, 2017-05-15:
Thanks for your suggestions. Overnight I also thought it might be some kind of qstr building issue. If the qstr's are collected before `frozen_mpy.c`, then the number of qstr's won't be correct. But I will follow up on your vague memory; I'll turn LTO off and see what happens. I will also look at the RAM usage as you suggest. And strings and names are significant part of some modules: I might try to trim there as well. I could create a minimal Keycode class that only defines the constants I need, for instance.

I'll follow up if/when I get some solutions.

dhalbert, 2017-05-23:
@tannewt I am going to do some measurements of stack usage to see how much I can safely grow the heap. I just want to check my assumptions with you.

If I understand correectly, stack grows downward from `&_estack` toward the top of the data (bss) section. In the (hypothetical wrong) case of 1000 words of RAM, say we have:
10:  bss start
500: bss end
501: stack limit
990: &_estack

The heap is allocated statically in the bss section. `&_estack` marks the stack base (I am reluctant to say "top" or "bottom", since it varies depending on whether you are talking physically or logically).

Is this right?
Is there an available global marking bss end or size or the lower stack limit?

 Thanks.

tannewt, 2017-05-23:
Hi @dhalbert, that sounds like something super useful. Your mental model matches mine.

The end of bss is marked by _ebss and _ezero [from the linker script](https://github.com/adafruit/circuitpython/blob/master/atmel-samd/boards/samd21x18-bootloader-external-flash.ld#L69). We use it to protect against infinite recursion in the python [here](https://github.com/adafruit/circuitpython/blob/master/atmel-samd/main.c#L568). We also reference the current stack size every time the GC runs [here](https://github.com/adafruit/circuitpython/blob/master/atmel-samd/main.c#L664).

I know that MicroPython on the Micro:bit uses a 2k stack so we should be able to shrink ours a bit. There may also be static data structures we can shrink as well.

Thanks for looking into this! It'll be super useful.

tannewt, 2017-05-24:
What is left to do on this issue versus #142?

dhalbert, 2017-05-24:
The original bug is that frozen modules won't build, which is still the case. You could close it as "won't fix", or leave it open for later. Or I can open a new one with the original details and discard all the RAM usage comments.

dhalbert, 2017-07-15:
#169 will fix the problems mentioned in the original post. The warnings that caused errors are made just warnings, and mpy-tool.py is called with the correct options.

@tannewt Your memory of a problem with "an array at the end of the struct" was correct. I've submitted a gcc bug report about this: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=81440 ("-Wlto-type-mismatch warning with flexible array in struct").

I'm successfully building and using images with a number of frozen modules. I'm hoping this will solve my RAM overflow issues.

closed 2017-07-17 by dhalbert