adafruit/circuitpython #134: esp8266 busio pin check error

jerryneedell, 2017-05-21:
on an esp8266 (Huzzah breakout) When initializing busio.I2C it returns an error for Pin DVDD "in use"

`from board import *
import busio
bus = busio.I2C(SCL,SDA)
`
generates:
ValueError: Pin VDD in use

tannewt, 2017-05-23:
Hi @jerryneedell, I can't reproduce this on the feather version. I'm using 0.10.1.

```
Adafruit CircuitPython 0.10.1 on 2017-05-20; ESP module with ESP8266
>>> from board import *
>>> import busio
>>> bus = busio.I2C(SCL, SDA)
>>>
>>>
PYB: soft reboot
#5 ets_task(40100164, 3, 3fff839c, 4)
boot.py output:

Press any key to enter the REPL. Use CTRL-D to soft reset.

Adafruit CircuitPython 0.10.1 on 2017-05-20; ESP module with ESP8266
>>> from board import *
>>> import busio
>>> bus - busio.I2C(SCL, SDA)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'bus' is not defined
>>> bus = busio.I2C(SCL, SDA)
>>>
```

Any thing I missed? Does it still repro for you?

jerryneedell, 2017-05-23:
Odd -  I was using a "non-feather" ESP8266 Huzzah - in case that makes a difference. I try both when I get home later this afternoon.

jerryneedell, 2017-05-23:
I guess you fixed it ... ;-)  I can't reproduce it now either. Sorry ... Feel free to close it, I'll repost it if/when I can reproduce it. - I hate problems that just go away. They usually reappear at the worst possible time ...

tannewt, 2017-05-23:
No worries! I'd rather have an issue filed that goes away versus not hearing about something at all. Thanks again @jerryneedell! Reopen if it happens again.

closed 2017-05-23 by tannewt