adafruit/circuitpython #141: Oddities with soft reboot and REPL history mechanism

dhalbert, 2017-05-24:
The command history seems to be preserved across ctrl-D soft reboot, but there appears to be junk in it:

```Adafruit CircuitPython 0.10.1-3-g70d0425 on 2017-05-21; Adafruit Feather M0 Express with samd21g18
>>> import gc
>>> import gc        [pressed up-arrow]
>>>                        [control-D]
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 0.10.1-3-g70d0425 on 2017-05-21; Adafruit Feather M0 Express with samd21g18
>>> import gc        [pressed up-arrow]
>>>0³        [pressed up-arrow again]
```

DIfferent weirdness after soft reboot with different input
```
>>> 2+3
5
>>>2+3
5
>>2+3               [Yes, that's one less >]
5
>>>
```

tannewt, 2017-05-24:
Good catch! Will look soon.

jerryneedell, 2017-05-24:
I have noticed this too, but it seems to be intermittent if it is saved and how well it is preserved. Sometimes it is there and sometimes not.

tannewt, 2017-05-24:
Looks like the mp_reset function is missing `readline_init0`. I'll add it and test tomorrow.

tannewt, 2017-05-24:
Thanks for the bug! Fixed it.

commit 3cad7b7b (2017-05-24): atmel-samd: correctly reset the REPL history on reload. Fixes #141
closed 2017-05-24 by tannewt