adafruit/circuitpython #142: atmel-samd: RAM usage investigations

dhalbert, 2017-05-24:
I'm splitting this off from #130 to document monitoring and allocating the available RAM in the atmel-samd ports.

dhalbert, 2017-05-24:
@tannewt: https://github.com/adafruit/circuitpython/issues/130#issuecomment-301558151:

> I'd probably take the opposite approach of trying to reduce RAM usage. There are a couple ways I could look into for that:
>-Increasing the heap size (just a statically allocated array) at the expense of the stack space.
>-Forcing the SPI flash code to use the scratch sector rather than a sector in memory, which is 4k. This will slow it down though.
>-Staring into the abyss of the heap tracking output and coming up with another idea. :-)

dhalbert, 2017-05-24:
@dhalbert: https://github.com/adafruit/circuitpython/issues/130#issuecomment-303395106:

> @tannewt I am going to do some measurements of stack usage to see how much I can safely grow the heap. I just want to check my assumptions with you.
> 
> If I understand correctly, the stack grows downward from &_estack toward the top of the data (bss) section. In the (hypothetical wrong) case of 1000 words of RAM, say we have:
> 10: bss start
> 500: bss end
> 501: stack limit
> 990: &_estack
> 
> The heap is allocated statically in the bss section. &_estack marks the stack base (I am reluctant to say "top" or "bottom", since it depends on whether you are talking physically or logically).
> 
> Is this right?
> Is there an available global marking bss end or size or the lower stack limit?
> 
> Thanks.

dhalbert, 2017-05-24:
@tannewt: https://github.com/adafruit/circuitpython/issues/130#issuecomment-303446616

> Hi @dhalbert, that sounds like something super useful. Your mental model matches mine.
> 
> The end of bss is marked by _ebss and _ezero from the linker script. We use it to protect against infinite recursion in the python here. We also reference the current stack size every time the GC runs here.
> 
> I know that MicroPython on the Micro:bit uses a 2k stack so we should be able to shrink ours a bit. There may also be static data structures we can shrink as well.
> 
> Thanks for looking into this! It'll be super useful.
>

dhalbert, 2017-05-24:
I added `micropython.max_stack_use()` which returns the maximum stack excursion since the last reset. This is in a branch: https://github.com/dhalbert/circuitpython/tree/stack_usage.
The basic idea, not original with me, is to fill the stack space with a marker value (I used 0xee) on startup. Use of the stack overwrites this value. To find the max stack watermark, we just scan from the stack ceiling towards the stack base upwards in RAM until we stop finding marker values.

Some preliminary results below. The available stack space is not quite 10k. The simple `foo(n)` function piles up `n` Python stack frames. Each Python call frame is 256 bytes. I then imported about 120 lines of code and ran it. The compiler used a lot of stack; actually executing the code did not exceed that high watermark. I'll try a lot more, but you get the idea.

```
>>> import micropython
>>> micropython.mem_info()
stack: 536 out of 9696
GC: total: 16000, used: 304, free: 15696
 No. of 1-blocks: 7, 2-blocks: 6, max blk sz: 2, max free sz: 972
>>> micropython.max_stack_use()
992
>>> def foo(n): return foo(n-1) if n > 0 else micropython.stack_use()
... 
>>> foo(0)
752
>>> foo(1)
1008
>>> foo(2)
1264
>>> micropython.max_stack_use()
1376
>>> from mappings import Mappings
>>> micropython.max_stack_use()
2672
>>> m = Mappings("switches.txt")
>>> micropython.max_stack_use()
2672
>>> 
```

tannewt, 2017-05-24:
Interesting! I'd be happy to put this into master. I'm am using _ezero for the safe mode stuff though. https://github.com/adafruit/circuitpython/blob/master/atmel-samd/main.c#L476

dhalbert, 2017-05-24:
Good point. I chose the stack limit, but that allows a 1k overflow area. So I need to start at just above _ezero.

I have this [hack](https://github.com/dhalbert/circuitpython/blob/stack_usage/atmel-samd/main.c#L574) in the code to avoid damaging the current frame.
`p = &here - 100;  // Don't smash stuff in current frame, allocated below here.`
Do you know if I can find out where the current frame ends (lowest used numerical address, aka "top" of frame)? I could call another routine and have it return the address of a variable declared by it, but I have to make sure the optimizer doesn't inline that or otherwise screw it up.

tannewt, 2017-05-24:
Thats the hack I know of. I'm sure theres a way to read the `sp` register that holds it but I'm not sure how off the top of my head.

dhalbert, 2017-08-07:
Closing this, covered by #175 and #182 (max stack usage measurement), #183 (add 4k to heap). Further investigations could get a new issue.

closed 2017-08-07 by dhalbert