adafruit/circuitpython #144: Stack overflow checking broken

dhalbert, 2017-06-01:
Stack overflow checking broke between 0.9.5 and 0.9.6.
```
>>> def r(n): return n if n==0 else r(n-1)
>>> r(20)
0
>>> r(50)
```
`r(50)` just hangs on the broken builds, but gives `RuntimeError: maximum recursion depth exceeded` on the good builds.

I did a bisect and narrowed it down to:
```
commit 8d0d5defc7e06143cc5cfc4599bbf10c1055c623
Author: Scott Shawcroft <scott.shawcroft@gmail.com>
Date:   Mon Apr 17 17:20:05 2017 -0700

    atmel-samd: Use LTO for all builds to get 5000 bytes back on express boards.
```

Possible cause(?):
I had trouble in my `max_stack_use()` code with automatic inlining of a function that reads the address of a local to determine a stack boundary. I made it `MP_NOINLINE`. That worked in a debug build but not with `-flto`.  For this bug, I tried `making mp_stack_ctrl_init()` in `py/stackctrl.c` be `MP_NOINLINE` but that did not help (but LTO is on). But this may be a red herring. Too tired to look further.

tannewt, 2017-06-01:
Thanks for letting me know! I should figure out how to read the stack pointer directly.

dhalbert, 2017-06-01:
I am not sure getting the stack pointer is really the problem here, but I successfully read the stack pointer directly by declaring:
```
volatile register char* sp asm("sp");`
```
That aliases `sp` to the actual register. Not sure if the `volatile` is necessary, but it may avoid some bad optimization that would break things. 

I originally used `asm("sp")` in py/stackctrl.h for some `max_stack_use()` routines, but that was not architecture-independent. So I took the address of a local instead. py/stackctrl.h does the same. It's possible -flto is messing this up somehow. From py/stackctrl.h:
```
void mp_stack_ctrl_init(void) {
    volatile int stack_dummy;
    MP_STATE_THREAD(stack_top) = (char*)&stack_dummy;
}
```
Maybe there should instead be a architecture-dependent routine or macro to fetch the stack pointer that gets used by py/stackctrl.h.

But as I said, I'm not actually sure this is the problem that's causing the bug. As you mentioned a few days, looking at the linker map would help. And I haven't tried gdb debugging with both `-g` and `-flto`.

**A side note:**

I tried to write a helper routine to just return a value like that but was stymied by the gcc warning/error that I was returning the address of a local. No amount of casting to and fro or `#pragma` warning suppression would get around it. I briefly got it to work by assigning it to a `uintptr_t`, but then when I compiled with -flto it got caught again as an error (if I remember right).
```
char* get_an_sp() {
    char dummy;
    return &dummy - 1;   // gcc will complain you are returning the address of a local.
}
```

dhalbert, 2017-06-01:
Doing a little research on LTO, found this Stack Overflow answer: https://stackoverflow.com/a/25649861/142996

>BTW, notice that the same optimization flags (-flto -Os) should be passed both at compile time and at link time. If you forgot -like you did- any of them at linking phase, LTO won't work! (When using make you want CC=gcc -flto -Os not CFLAGS= -flto -Os).

In the atmel-samd/Makefile, I see `-Os' on `CFLAGS` but not `LDFLAGS`. On the other hand, it does seem to work.

tannewt, 2017-06-01:
LDFLAGS does use it because it includes all of the CFLAGS: https://github.com/adafruit/circuitpython/blob/master/atmel-samd/Makefile#L153

Good thought though! I expect it is due to variable reordering or optimization. Thats why reading the $sp register directly would be nice.

dhalbert, 2017-06-02:
Some more info, built on the latest git pull from master.

1. Just to confirm, turning off `-flto` restores proper stack checking.
2. I changed the routines in py/stackctrl.c to use $sp to find the lowest point in the stack, and that made no difference. Neither did declaring them `MP_NOINLINE`. I undid those changes for next test.
3. It appears that [`mp_stack_check()`](https://github.com/adafruit/circuitpython/blob/master/py/stackctrl.c#L58) is not being called or is not doing its job. Here's a fancier version of the test program:
```
import micropython
def r(n):
    if n == 0:
        return 0
    else:
        micropython.mem_info()
        r(n-1)
```
`mem_info()` prints out `mp_stack_usage()` and `MP_STATE_THREAD(stack_limit)`, which are exactly the values used by `mp_stack_check()`. But If I call `r(50)` the stack goes way beyond the limit. I get  this output:
```
...
stack: 10512 out of 9696
GC: total: 16000, used: 1088, free: 14912
 No. of 1-blocks: 23, 2-blocks: 10, max blk sz: 8, max free sz: 902
stack: 10768 out of 9696
GC: total: 16000, used: 10    [hangs here; presumably after smashing something]
```
10768 is beyond the [1k safe zone defined in main.c.](https://github.com/adafruit/circuitpython/blob/master/atmel-samd/main.c#L571)

If LTO is off, I get more expected output; the overflow gets noticed in the safe zone:
```
stack: 9488 out of 9648
GC: total: 15936, used: 1184, free: 14752
 No. of 1-blocks: 27, 2-blocks: 11, max blk sz: 8, max free sz: 898
stack: 9768 out of 9648
GC: total: 15936, used: 1184, free: 14752
 No. of 1-blocks: 27, 2-blocks: 11, max blk sz: 8, max free sz: 898
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 6, in r
[...]
RuntimeError: maximum recursion depth exceeded
```
So this seems like LTO messing up the calls to MP_STACK_CHECK() in various places in the code. Not sure how, and time for sleep!

closed 2017-06-05 by tannewt

dhalbert, 2017-06-05:
This was a Heisenbug. I put a print statement in `mp_stack_check()` and the problem went away. So I figured it was some kind of inlining issue.

tannewt, 2017-06-05:
:-) Thanks for figuring it out!

PR #146 (2017-06-05): Fix broken mp_stack_check() when using -flto
Fixes #144. Force `py/stackctrl.c` routines that effectively read the $sp not be to inlined, by using `asm("")`. `MP_NOINLINE` is not effective to prevent inlining in this case. Something about `-flto` makes these routines not work correctly. They work fine without `-flto`.

With this fix, `MP_STACK_CHECK()` stack overflow checking works again with `-flto`.