adafruit/circuitpython #145: Support main.mpy/code.mpy files to execute at startup.

tdicola, 2017-06-03:
It looks like right now main code execute at boot has to be a pure python .py file.  This is tricky though since pure python code has a significant memory overhead during parsing.  Beyond about 100 lines of code in a main.py it becomes impossible to import a few modules and run code.  An easy way around this is switching to .mpy files that are processed ahead of time to reduce their size and parsing memory usage.  Let's see if we can support main.mpy/code.mpy files execute on boot for those cases where users need the most memory possible.

tannewt, 2017-06-05:
Would you mind posting the main.py file? I'm hesitant to support mpy'd mains because they are opaque files. I'd rather have it always possible to read main and instead factor out stuff into mpy'd libraries.

tannewt, 2017-06-07:
I believe this is less of an issue now. I believe @tdicola was using ampy to load main.py which copies all of the file contents into memory before parsing. Saving main.py to the internal storage first will mitigate this.

closed 2017-06-07 by tannewt

sharifrazzaque, 2025-12-28:
This is a very old issue, but I'm wondering if having the main as a mpy would **reduce the time to from power-on to starting the script.**  It currently takes 8 second to go from power-on to running a tiny main.py which just turns on the LED (on a Adafruit ESP32 Feather V2).  This is too long for my purpose. The alternative appears to port my app to compiled Ardunio C++?

dhalbert, 2025-12-28:
@sharifrazzaque We have some open issues and PR's about this kind of thing, and I am planning to add some options to settings.toml to allow turning off or varying some of these delays.

bablokb, 2025-12-28:
@sharifrazzaque Removing the wifi credentials from settings.toml is the one single thing that gains you the most time during boot. There are other things to do as well, but this is a really low hanging fruit.

sharifrazzaque, 2025-12-28:
Thanks @bablokb! This cut my boot time from 8 to 4.5 seconds.