adafruit/circuitpython #147: reverse() returns None

ladyada, 2017-06-05:
```
>>> ledpins = [0, 1, 2]
>>> ledpins.reverse() -> None
>>> ledpins[::-1] -> [2, 1, 0]
```
I prefer reverse() for readability, but for now am using [::-1]

dhalbert, 2017-06-05:
`reverse()` reverses a list in place, and deliberately does not return anything, to remind you it's not a function, but something with a side-effect.
```
>>> l = [1,2,3]
>>> l.reverse()
>>> l
[3, 2, 1]
```
If you want a functional operation, use `list.reversed()`. Scott recently took a pull request from me that turned that on in CircuitPython, as well as some other things: #138.

ladyada, 2017-06-05:
oh weird! i think maybe i didnt have reversed() and got a confused :)

closed 2017-06-05 by ladyada