adafruit/circuitpython #157: Win8 Mass Storage not showing [Feather M0 Express]

mindforger, 2017-06-21:
I followed the guide and the LED is fading, but there is no Mass Storage Device showing, only a COM Port! What's wrong?

tannewt, 2017-06-21:
Windows is funny. Have you tried connecting to the COM port via putty or similar programs? I've heard from others that that causes the drive to show up.

mindforger, 2017-06-21:
will try it

mindforger, 2017-06-22:
no success, unfortunately, thank you anyways ... 

PS: as my other issue got closed a bit boldly and i want to prevent an misunderstanding about this, **i do not want** to register myself with the forum, github must be enough to report this!

tannewt, 2017-06-22:
Sorry, I don't have a Windows 8 computer to try it with. Do you have any other computers you can try with?

mindforger, 2017-06-23:
This afternoon, in about 10 hours, will report back then

dhalbert, 2017-06-23:
@mindforger The very first M0 Express boards shipped had a buggy bootloader. Also, as @tannewt mentioned, the drivers can act oddly. See #123 for some similar symptoms. You might try these things:

1. Uninstall all the Adafruit drivers. That will probably make the COM port disappear See if FEATHERBOOT then appears when you double-click the reset button. If it does appear, see what's in `INFO_UF2.TXT`, which will report the version of the bootloader. If it's older than V1.21 or V1.22, install the V1.22 bootloader by dragging it on top of the FEATHERBOOT drive. See details [here](https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/uf2-bootloader#updating-the-bootloader).

2. If the FEATHERBOOT drive still does not appear, try rebooting and reinstalling the drivers from the latest Adafruit [Windows drivers package](https://github.com/adafruit/Adafruit_Windows_Drivers/releases). I have seen this behavior on Windows 7: the drivers were installed, FEATHERBOOT did not appear, I uninstalled and reinstalled the drivers, and then FEATHERBOOT appeared consistently.

mindforger, 2017-06-23:
hahahaha win 10 is working and it is 1.18 .. well that explains a lot :)

closed 2017-06-23 by mindforger

tannewt, 2017-06-26:
@mindforger does updating the bootloader fix Windows 8 for you?

mindforger, 2017-06-26:
Oh well, forgot to test it with all those work, will Test tomorrow

tannewt, 2017-06-26:
Thanks! I'd expect it to work but would love confirmation.

mindforger, 2017-06-27:
working now, afte removing the Adafruit COM Driver :)