adafruit/circuitpython #159: UART gives OSerror when you try to use it?

ladyada, 2017-07-02:
> main.py output:
> Traceback (most recent call last):
>   File "main.py", line 14, in <module>
> OSError: 11

when running following (error on uart.read())

```
# Gemma IO demo #7 - USB/Serial echo

from digitalio import *
from board import *
import busio
import time

led = DigitalInOut(D13)
led.direction = Direction.OUTPUT

uart = busio.UART(D0, D2, baudrate=9600)

while True:
    d = uart.read()
    if d != None:
        led.value = True
        print(d)
        led.value = False
```

hukuzatuna, 2017-07-02:
Bug replicated perfectly on Metro M0 Express running CircuitPython 1.0.0-rc.2.

tannewt, 2017-07-03:
OSError: 11 Is telling you to try again. You could then catch the exception with `try` `except`. However, it looks like most ports switch the mode to non-blocking mode where it would return `None` instead of an exception. I'll fix this up and close the issue with it.

`read` will read until the end of file so I recommend giving it a limit instead like `read(1)`.

Another option that works will with GPS is:

```python
# Gemma IO demo #7 - USB/Serial echo

from digitalio import *
from board import *
import busio
import time

led = DigitalInOut(D13)
led.direction = Direction.OUTPUT

uart = busio.UART(TX, RX, baudrate=9600)

while True:
    d = uart.readline()
    if d != None:
        led.value = True
        print(d)
        led.value = False
```

Output is:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
b'br\xb11.45,K,A*3E\r\n'
b'$GPGGA,185934.000,4740.4632,N,12223.0595,W,1,04,2.32,167.3,M,-17.3,M,,*5D\r\n'
b'$GPGSA,A,3,28,17,19,06,,,,,,,,,2.52,2.32,1.00*07\r\n'
b'$GPRMC,185934.000,A,4740.4632,N,12223.0595,W,0.66,277.67,030717,,,A*72\r\n'
b'$GPVTG,277.67,T,,M,0.66,N,1.22,K,A*3F\r\n'
b'$GPGGA,185935.000,4740.4632,N,12223.0590,W,1,04,2.32,167.3,M,-17.3,M,,*59\r\n'
b'$GPGSA,A,3,28,17,19,06,,,,,,,,,2.53,2.32,1.00*06\r\n'
b'$GPRMC,185935.000,A,4740.4632,N,12223.0590,W,0.50,281.30,030717,,,A*78\r\n'
b'$GPVTG,281.30,T,,M,0.50,N,0.93,K,A*3A\r\n'
```

willingc, 2017-07-03:
> OSError: 11 Is telling you to try again.

@tannewt Can the error message be expanded a bit (i.e. retry) so that the user knows how to resolve?

tannewt, 2017-07-03:
@willingc Yeah, that would be nice. Switching to returning None makes more sense though I think. There is some weirdness where we return OSError and error numbers to keep compatibility with CPython. It'd be good to get rid of them though.

closed 2017-07-03 by tannewt

willingc, 2017-07-03:
Agree on returning None since that covers more cases. Makes sense re: error numbers.

ladyada, 2017-07-03:
oh yah the 'standard' pyserial way of doing this is returning None - thanks! :)

commit 76d2c28e (2017-07-03): atmel-samd: Switch streams to non-blocking mode. Fixes #159.
commit e02e4235 (2017-07-12): atmel-samd: Switch streams to non-blocking mode. Fixes #159.