adafruit/circuitpython #162: Add ability to turn off status neopixel.

tannewt, 2017-07-10:
Add an API for turning off the status neopixel by default. Right now, it falls back to status messages when not being used by the user. This would cause it to turn off when not in use by the user.

Request comes from nis_ on the Adafruit forums: https://forums.adafruit.com/viewtopic.php?f=60&t=114694&p=599762#p599762

tannewt, 2017-08-21:
@asherlie Do you still have time to do this? If not, please unassign yourself. Thanks!

asherlie, 2017-09-05:
i added an implementation for this on my fork [here](https://github.com/asherlie/circuitpython/tree/dim_neopixel). i added 3 functions to the samd module for the user to control the behavior of the neopixel:
- enable_dim_neopixel()
- disable_dim_neopixel()
- set_dim_level()

enable_dim_neopixel and disable_dim_neopixel turn on and off dim neopixel mode. set_dim_level takes in an integer from 0-255 and sets the brightness level of dim neopixel mode. The default brightness level is 40.

tannewt, 2017-09-05:
Thanks @asherlie! Could you open a pull request with it so we can discuss it? Thanks!

commit 5aa89220 (2017-09-12): add set_rgb_status_brightness (#246)

Add set_rgb_status_brightness() via `samd.rgb_status_brightness`.

Fixes #162.
closed 2017-09-12 by tannewt